ldab/ESP32_FTPClient #4: If wifi connection drops it hangs cpu

spyder0069, 2019-08-08:
If ftp.OpenConnection(); is called when the wifi connection had dropped the process blocks/hangs the cpu.  

In my case I had it updating every 5 seconds and then I rebooted my router.  The serial debug shows:

Connecting to:  (ftp address)
Command connected

Then it halts.

ldab, 2019-08-09:
I will have a look at it later, thanks.

As a quick fix, you can add a timeout here:
https://github.com/ldab/ESP32_FTPClient/blob/45e7f02e3316dd91dfa38a90205d1724bbbb634c/src/ESP32_FTPClient.cpp#L42

Or check connection before calling for the FTP function? `WiFi.isConnected();`

spyder0069, 2019-08-09:
WiFi.isConnected(); doesn't noticed a dropped connection for 20 seconds so the program still has the problem.  Even when the router is back up and the program recovers the wifi the ftp section doesn't recover.  I am guessing that it breaks the client and it needs to be initialized again?  I'll have a chance to play around with it more later today.

spyder0069, 2019-08-09:
I've got things working.  Not sure if there is a more elegant way of doing things but in summary:

I changed the OpenConnection to include a timeout for the client.connect.  This way the user can select how long they want to block.  Its in milliseconds.

I created a variable called ftpError and anytime there is a problem that gets incremented and other subs check and skip their routine if there in an error.  The error is cleared on a OpenConnection. 

It does need to check for wifi at the beginning of a OpenConnection and if its not there if sets the error.  While this doesn't catch a recently dropped connection it keeps it from trying to do a client.connect a second time which will hang the cpu.  Here are my resulting files:

`class ESP32_FTPClient
{
  private:
  void WriteClientBuffered(WiFiClient* cli, unsigned char * data, int dataLength);
  char outBuf[128];
  char outCount;
  byte ftpError;
  word ftpTimeout;
  WiFiClient client;
  WiFiClient dclient;
  char* userName;
  char* passWord;
  char* serverAdress;
  unsigned char clientBuf[1500];
  size_t bufferSize = 1500;
  WiFiClient* GetDataClient();
  
  public:
  ESP32_FTPClient(char* _serverAdress, char* _userName, char* _passWord);
  void OpenConnection(word ftptimeout);
  void CloseConnection();
  void NewFile (char* fileName);
  void AppendFile( char* fileName);
  void WriteData (unsigned char * data, int dataLength);
  void CloseFile ();
  void GetFTPAnswer (char* result = NULL, int offsetStart = 0);
  void GetLastModifiedTime(char* fileName, char* result);
  void RenameFile(char* from, char* to);
  void Write(char * str);
  void InitFile(char* type);
};`

spyder0069, 2019-08-09:
`#include <WiFiClient.h>
#include <WiFi.h>
#include "ESP32_FTPClient.h"

ESP32_FTPClient::ESP32_FTPClient(char* _serverAdress, char* _userName, char* _passWord){
  userName = _userName;
  passWord = _passWord;
  serverAdress = _serverAdress;
}

WiFiClient* ESP32_FTPClient::GetDataClient() {
  return &dclient;
}

void ESP32_FTPClient::GetLastModifiedTime(char* fileName, char* result) {
  if(ftpError==0){
      Serial.println("Send MDTM");
      client.print(F("MDTM "));
      client.println(F(fileName));
      GetFTPAnswer (result, 4);
  }
}

void ESP32_FTPClient::WriteClientBuffered(WiFiClient* cli, unsigned char * data, int dataLength) {
  
  size_t clientCount = 0;
  for(int i = 0; i < dataLength;i++){
    clientBuf[clientCount] = data[i];
    //client.write(data[i])
    clientCount++;
    if (clientCount > bufferSize-1) {
        cli->write(clientBuf, bufferSize); 
      clientCount = 0;
    }
  }
  if (clientCount > 0){
      cli->write(clientBuf, clientCount); 
  }
}

void ESP32_FTPClient::GetFTPAnswer (char* result, int offsetStart) {
  byte respCode;
  byte thisByte;
  outCount = 0;
  ftpTimeout=0;
  while (!client.available()){
    delay(1);
    if(ftpTimeout<1000){
      ftpTimeout++;
    }else{
      ftpError++;
      break;
    }
  }

  //if(!client.available()){ftpError++;}

  if(ftpError==0){
      while (client.available()) {
        thisByte = client.read();
        if (outCount < 127) {
          outBuf[outCount] = thisByte;
          outCount++;
          outBuf[outCount] = 0;
        }
      }
      if(result != NULL){
        Serial.println("Result start");
        for(int i = offsetStart;i<21;i++){
          result[i] = outBuf[i - offsetStart];
        }
        Serial.print("Result: ");
        Serial.write(result);
        Serial.println("Result end");
      }
  }
}

void ESP32_FTPClient::WriteData (unsigned char * data, int dataLength) {
  if(ftpError==0){
      Serial.println(F("Writing"));
      WriteClientBuffered(&dclient, &data[0], dataLength);
  }
}

void ESP32_FTPClient::CloseFile () {
  if(ftpError==0){
      Serial.println(F("Close File"));
      dclient.stop();
      GetFTPAnswer();
  }
}

void ESP32_FTPClient::Write(char * str) {
  GetDataClient()->print(str);
}

void ESP32_FTPClient::CloseConnection() {
  client.println(F("QUIT"));
  client.stop();
  Serial.println(F("Connection closed"));
  Serial.println("ftpError=" + String(ftpError));
}

void ESP32_FTPClient::OpenConnection(word ftptimeout) {  
  if(WiFi.isConnected()){
      ftpError=0;
      Serial.print(F("Connecting to: "));
      Serial.println(serverAdress);      
      if (client.connect(serverAdress, 21, ftptimeout)) {  // 21 = FTP server //4000=4 second timeout
        Serial.println(F("Command connected"));
      }       
      GetFTPAnswer();
      
      if(ftpError==0){
      Serial.println("Send USER");
      client.print(F("USER "));
      client.println(F(userName));
      GetFTPAnswer();
      }
    
      if(ftpError==0){
      Serial.println("Send PASSWORD");
      client.print(F("PASS "));
      client.println(F(passWord));
      GetFTPAnswer();
      }
    
      if(ftpError==0){
      Serial.println("Send SYST");
      client.println(F("SYST"));
      GetFTPAnswer();
      }
  }else{
    ftpError=0;
  }
}

void ESP32_FTPClient::RenameFile(char* from, char* to) {
  if(ftpError==0){
  Serial.println("Send RNFR");
  client.print(F("RNFR "));
  client.println(F(from));
  GetFTPAnswer();
  }
  if(ftpError==0){
  Serial.println("Send RNTO");
  client.print(F("RNTO "));
  client.println(F(to));
  GetFTPAnswer();
  }
}

void ESP32_FTPClient::NewFile (char* fileName) {
  if(ftpError==0){
      Serial.println("Send STOR");
      client.print(F("STOR "));
      client.println(F(fileName));
      GetFTPAnswer();
  }
}

void ESP32_FTPClient::InitFile(char* type){
  if(ftpError==0){
      Serial.print("Send ");
      Serial.println(type);
      client.println(F(type));
      GetFTPAnswer();
  }
  if(ftpError==0){
      Serial.println("Send PASV");
      client.println(F("PASV"));
      GetFTPAnswer();
     
      Serial.println(outBuf);
      char *tStr = strtok(outBuf, "(,");
      int array_pasv[6];
      for ( int i = 0; i < 6; i++) {
        tStr = strtok(NULL, "(,");
        array_pasv[i] = atoi(tStr);
        if (tStr == NULL) {
          Serial.println(F("Bad PASV Answer"));
        }
      }
      unsigned int hiPort, loPort;
      hiPort = array_pasv[4] << 8;
      loPort = array_pasv[5] & 255;
    
      Serial.print(F("Data port: "));
      hiPort = hiPort | loPort;
      Serial.println(hiPort);
      if (dclient.connect(serverAdress, hiPort)) {
        Serial.println(F("Data connection etablished"));
      }
  }
}

void ESP32_FTPClient::AppendFile (char* fileName) {
  if(ftpError==0){
      Serial.println("Send APPE");
      client.print(F("APPE "));
      client.println(F(fileName));
      GetFTPAnswer();
  }
}`

spyder0069, 2019-08-09:
Example of new usage:
  ftp.OpenConnection(4000);  //4 second timeout.  timeout is in ms
  ftp.InitFile("Type A");
  ftp.NewFile("hello_world.txt");
  ftp.Write("Hello World");
  ftp.CloseFile();

I will attach the files to this one.  :^)

[ESP32_FTPClient.zip](https://github.com/ldab/ESP32_FTPClient/files/3487074/ESP32_FTPClient.zip)

spyder0069, 2019-08-09:
I did get it to crash if the router reset mid ftp.  The decode shows:

0x400d1bbe: ESP32_FTPClient::InitFile(char*) at C:\Users\user\AppData\Local\Temp\arduino_build_413665\sketch\ESP32_FTPClient.cpp line 181
0x400d1c8f: mynewfile() at C:\SPYDERROBOTICSDATA\spyderrobotics\FIRMWARE\arduino\SKETCHBOOK\FTP_TEST\test4\ESP32_FTPClient/ESP32_FTPClient.ino line 64
0x400d1cc5: loop() at C:\SPYDERROBOTICSDATA\spyderrobotics\FIRMWARE\arduino\SKETCHBOOK\FTP_TEST\test4\ESP32_FTPClient/ESP32_FTPClient.ino line 34
0x400d3f89: loopTask(void*) at C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3-rc1\cores\esp32\main.cpp line 19
0x400885dd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

I think its crashing at the same line as when the password was wrong.  Not sure how to correct that.

spyder0069, 2019-08-09:
Seemed to have fixed it.  I needed one more error check in the InitFile.  Here are the new files.
[ESP32_FTPClient.zip](https://github.com/ldab/ESP32_FTPClient/files/3487157/ESP32_FTPClient.zip)

ldab, 2019-08-09:
> WiFi.isConnected(); doesn't noticed a dropped connection for 20 seconds so the program still has the problem. Even when the router is back up and the program recovers the wifi the ftp section [...]

This implementation needs to be in your code, not in the library.

> I think its crashing at the same line as when the password was wrong. Not sure how to correct that.

It's not, it crashed on `ESP32_FTPClient.cpp line 181` at function `InitFile()` but I'm unable to help since I don't know what is on line 181 of your file.

I will consider your comments, but it's not a very elegant way of fixing it, I may work on it next week.

spyder0069, 2019-08-09:
at the time of the crash line 181 was:

array_pasv[i] = atoi(tStr);

Which was the same line that cause the crash/restart when the password was wrong.  I suspect in this case the internet dropped during the transfer/receive of that sub and then recreated the error because it was incomplete or corrupt. 

As far as the checking for wificonnection in library is concerned:

Is there some reason to not do this in the library?  Doesn't the entire function depend on connected wifi?  Just trying to understand why this would be bad.  :^)

ldab, 2019-08-09:
I just noticed you're using a different library than the latest one on this repository, I have implemented few extra functions.

> Is there some reason to not do this in the library? Doesn't the entire function depend on connected wifi? Just trying to understand why this would be bad. :^)

Because you start to rely on another library, and the idea is keep it simple and potentially compatible cross platforms.

spyder0069, 2019-08-09:
Ah, maybe some of these have already been fixed.  I didn't realize this page was a fork of the code.  My apologies.  I will try the new version and see how things work.  :^)

ldab, 2019-08-09:
You can now define a timeout on the constructor `ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, Timeout);` default is 10sec.

Code added timeout after every command when no internet connection.

126e0a118fa1f02929ad639205d1b6bc937b9809

closed 2019-08-09 by ldab