ldab/ESP32_FTPClient #5: hanging when ftp password is incorrect

spyder0069, 2019-08-09:
With the current code from this repository if a password is wrong it has a long pause when it returns the dataport number but does continue and when it gets to the Closefile the last thing it tries to run is GetFTPAnswer(); but does not return from that.

`void ESP32_FTPClient::CloseFile () {
  Serial.println(F("Close File"));
  dclient.stop();
  Serial.println("still running1");
  GetFTPAnswer();
  Serial.println("still running2");
}`

I do not get a serial output afterwards and it never returns back to the main loop.

Also ideally if the pasv is bad then it should exit instead of trying to connect and give the long pause for the data connection. 
 
Maybe Initfile can  return a response whether it was successfull so the user app can then determine if it needs to continue?

Plus I am not sure why it responds with data connection established when the password fails?

spyder0069, 2019-08-09:
Also there is a spelling error in established.  :^)

Serial.println(F("Data connection etablished"));

ldab, 2019-08-09:
> With the current code from this repository if a password is wrong it has a long pause when it returns the dataport number but does continue and when it gets to the Closefile the last thing it tries to run is GetFTPAnswer(); but does not return from that.

This should be fixed with the latest commit, 126e0a118fa1f02929ad639205d1b6bc937b9809 same problem which cause CPU to hang when Offline.

> Maybe Initfile can return a response whether it was successful so the user app can then determine if it needs to continue?

Agreed, and I plan to do that at some point, but for now consider it as a patch.

> Plus I am not sure why it responds with data connection established when the password fails?

Because `connection established` refers to the client connection and not the server response referring to the password, but yes, misleading and should be fixed when the above issue is fixed.

*I will keep this issue open to remind me to return a `bool` or something, check password and connection*

spyder0069, 2019-08-09:
New code with the timeout no longer hangs.  Thank you for your time and effort!

ldab, 2019-08-10:
check latest commit, bd6cf40ccb12eb386604e61051043f49241841de

Last argument, start the debbug, 2 = everything, 1 = error, 0 = nothing.
`ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);`

ldab, 2019-08-10:
PS. The function keeps running when Offline. will fix it tomorrow.

ldab, 2019-08-11:
0ddad8aa0a617c37d16d81de6ffd2c6f52ca23e8

closed 2019-08-11 by ldab