ldab/ESP32_FTPClient #6: hanging when connection is dropped

spyder0069, 2019-08-09:
I don't know if this was on your radar but when the connection is dropped I am getting a hang when it calls 

if (dclient.connect(serverAdress, hiPort)) {

Maybe this doesn't have a timeout?  Or is the data corrupt.  The serial output looks like:

Send PASV
Bad PASV Answer
Data port: 4228875273
(then hangs)

That port number is very high.   Is that because the response got corrupted and maybe that is hanging the connect because the port number is out of bounds or something?

ldab, 2019-08-10:
I can't replicate your issue, i dropped my WiFi after connecting and the code goes thru:
```
Send SYST
Send Type A
Send PASV
Bad PASV Answer
Data port: 4241949696
[E][WiFiGeneric.cpp:652] hostByName(): DNS Failed for files.000webhost.com
After initiFile() the loop
```

> That port number is very high. Is that because the response got corrupted and maybe that is hanging the connect because the port number is out of bounds or something? Maybe if the port number is not a valid range it needs to exit.

The Port number is probably just a random number as the array hasn't been initialized. But this will be fixed when the connection confirmation is implemented.

spyder0069, 2019-08-10:
Am I somehow not getting your latest code?  I updated and still get the problem.  I will attach my files.  I get to the data port and it just hangs. 

[test5.zip](https://github.com/ldab/ESP32_FTPClient/files/3488840/test5.zip)

spyder0069, 2019-08-10:
I usually wait until it starts the connection and immediately reboot the router.  Here is video of my results with the code I attached:  https://youtu.be/E4CszPF_3MU

ldab, 2019-08-10:
Please check against the last commit

spyder0069, 2019-08-10:
That was with the latest code.  I noticed I can recreate the hang on the data port without touching the router if I just call the wifi disconnect after I do the ftpopen:

`  ftp.OpenConnection();  //4 second timeout.  timeout is in ms
  WiFi.disconnect();
  ftp.InitFile("Type A");
  ftp.NewFile("hello_world.txt");
  ftp.Write("Hello World2");
  ftp.CloseFile();`

See if that recreates for you.

spyder0069, 2019-08-10:
One way to fix it is in the InitFile if there is a bad pasv answer (which there is when the connection is dropped) instead of breaking out of the for statement just do a return.  Here is an example:

`    if (tStr == NULL) {
      FTPdbgn(F("Bad PASV Answer"));
      //break;
      return;
    }`

Do you see any harm from that?

spyder0069, 2019-08-10:
Figure there is no need to try to do the dclient.connect if it didn't pass the pasv.  Seems like a reasonable solution.

spyder0069, 2019-08-10:
Also it doesn't seem like the isConnected is working.  Seems to still giving the timeout pause for each of the commands after the disconnect of wifi instead of bypassing and returning.  Looking more into that now.

spyder0069, 2019-08-10:
The solution to the isconnected always giving the 3 second timeout is in the GetFTPAnswer there is a while loop waiting for the client to become available (with the timeout).  After that loop though the outBuf isn't going to get any data and so the isconnected never returns false there.  So what I did was after the while loop that waits for the client to be available I just added a condition that if the client is still not available I set your _isConnected to false, call your sub and then return out.  In this scenario you get the initial pause while its trying to connected but then bypasses every other command because the isconnected is false.  One other thing I did was in the OpenConnection after every GetFTPAnswer call I check the isconnected and return if false.  This way as soon as a connection problem exists it exits that sub.

spyder0069, 2019-08-10:
One thing you might want to address with my mod is when you exit the GetFTPAnswer after the timeout since its exiting before the outBuf gets any data you don't exactly have a clear serial debug error message.  I'll post my modified cpp in a moment.

spyder0069, 2019-08-10:
I will attach my current cpp.  So after the GetFTPAnswer was modified I went through all the subs and any sub that makes a call to GetFTPAnswer and then has additional code after that call I put a check for the isConnected right after the GetFTPAnswer call.  This way it checks at the beginning of every sub for the connection and then also before proceeding with anycode that relies on the answer from the GetFTPAnswer sub.  This also eliminates additional wasted time trying to connect when it  the connection has been dropped.  

[ESP32_FTPClient.zip](https://github.com/ldab/ESP32_FTPClient/files/3489373/ESP32_FTPClient.zip)

ldab, 2019-08-11:
I could not reproduce using the following, the code continues, but I believe I am using a different version that you are, Are you downloading it from here or from Arduino or PlatformIO library manager?
```
  ftp.OpenConnection();
  WiFi.disconnect();
```

Please, keep your comments more concise, it's very hard to keep track of all comments, and codes, etc...

Also, I'm not downloading the *.zip and comparing it to mine, if you want to contribute with your code, please share it here properly formatted, highlighting differences. Alternatively, GitHub has it built-in, called `Pull-request`, it's way easier that way, and less confusing to me.

I have made a latest commit which will probably fix the problems mentioned, please try with it 0ddad8aa0a617c37d16d81de6ffd2c6f52ca23e8

If you're using Library manager, I pushed the new version, updates are overnight, latest release is v0.1.0

spyder0069, 2019-08-11:
I go to the code page on this github and download the library in a .zip.  When you make a change is that the correct way for me to get the latest?

ldab, 2019-08-11:
Yes, that should be it, please try with the latest one and let me know.

spyder0069, 2019-08-11:
The latest still has problems and hangs with my example.  I have only downloaded from github before so I am new to this process.  I did a fork and added my changes and then did a pull request.  (This is all new lingo to me).  Looks like it outlines my changes.  Hopefully I did it right.  This updated code works.  Let me know what you think about closing the connection after the bad pasv request.

ldab, 2019-08-11:
I can't reproduce your hang. Please paste your code here so that I can try

spyder0069, 2019-08-11:
`#include <WiFi.h>
#include <WiFiClient.h>    
#include "ESP32_FTPClient.h"

#define ftp_server        ""
#define ftp_username      ""
#define ftp_password      ""
#define ftp_timeout 3000
#define ftp_verbose 2

#define wifi_ssid         ""
#define wifi_password     ""

ESP32_FTPClient ftp (ftp_server,ftp_username,ftp_password,ftp_timeout,ftp_verbose);

void setup() {
  Serial.begin(115200);
  WiFi.begin(wifi_ssid, wifi_password);
  
  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.println("Connecting to WiFi..");
      
  }
  Serial.println("IP address: ");
      
  Serial.println(WiFi.localIP()); 
}

void loop() {
  mynewfile();
  Serial.println("Looping=" + String(WiFi.isConnected()));
  delay(3000);
}

void mynewfile(){ 
  ftp.OpenConnection();  
  WiFi.disconnect();
  ftp.InitFile("Type A");
  ftp.NewFile("hello_world.txt");
  ftp.Write("Hello World2");
  ftp.CloseFile();
}`

spyder0069, 2019-08-11:
The above will recreate the problem where it hangs on the data port.  The serial results are:

Command connected
Send USER
Send PASSWORD
Send SYST
Type A
FTP error: Offline
Send PASV
FTP error: Offline
Bad PASV Answer
Data port: 4228875414

spyder0069, 2019-08-11:
One thing to note that even though the wifi disconnect was called directly after creating the client connection you can see the user/password/syst makes it through.  As if the wifi disconnect doesn't affect the client.available right away.  Then it makes it to the pasv part and tries to connect to the bad port info and hangs.

ldab, 2019-08-11:
47d982fe477ffd32f70bcac02ded98e522d7f513 Done

closed 2019-08-11 by ldab