ldab/ESP32_FTPClient #8: ftp.write

mehrdad987, 2019-08-26:
it does not write files larger than 63K 
and  the semi-finished file is released
Are there limits defined for this part FTP.write

ldab, 2019-08-26:
Is it a question or a statement?

It's not clear to me your comment. Can you please post an example?

Do you have enough heap memory to hold 64k of data?

ldab, 2019-08-27:
700K of String??

Anyway, clearly you don't have enough free memory to allocate the SD card content.

You can check that by 'ESP.getMaxAllocHeap()'

closed 2019-08-27 by ldab

mehrdad987, 2019-08-27:
sorry for my bad explanation. I have to send about 700k file (that's come from sd) to my server but, it only moves up to 63k and no more data is sent

  File fie = SD.open(temp);
  String jib = fie.readString();
  ftp.Write(jiboty.c_str());
  ftp.CloseFile();

KvarTechnologies, 2023-03-27:
How did u then send the file ? @ldab