ldab/ESP32_FTPClient #10: It is possible to send data to SPIFF

anctsys, 2019-12-23:
Hi

It is possible to send data saved into SPIFF to server.
Can you post an example please

Thanks

Bruno

ldab, 2019-12-23:
Hi, yes it is possible.

I don't have an example, I will leave the issue open and if I have time in the future can work on it.

What is the use case? Which spiffs library do you use?

anctsys, 2019-12-23:
Hi 

I using ESP32 WROOM32 and my use case is to send photo capture to my ftp server.

I install this tools to manage file on SPIFF : 
https://github.com/me-no-dev/arduino-esp32fs-plugin/releases/
from https://randomnerdtutorials.com/esp32-troubleshooting-guide/

In the future i would use an ESP32 CAM to do it.

Bruno

anctsys, 2019-12-24:
Hi 
I add a new function to send stream (file from SPIFF and other) to ftp server .
I add an new method void ESP32_FTPClient::WriteData (Stream * stream), i add also an example and a jpeg to test this new function.
You can test it and add it to you library.

ref : https://github.com/anctsys/ESP32_FTPClient

I new in c++ , if you have remark, you can send to me.

Have a nice day

Bruno

closed 2019-12-26 by ldab