ldab/ESP32_FTPClient #13: SPIFFS

Vicentetomas, 2020-03-22:
Hi
I try to execute de Spfiss code.
I find a problem in this line: ftp.WriteData(&file);
.Can you please help me?. Thank you very much

Arduino:1.8.10 (Windows 10), Tarjeta:"ESP32 Wrover Module, Huge APP (3MB No OTA/1MB SPIFFS), QIO, 80MHz, 115200, None"

C:\Users\rafa\Documents\Arduino\sketch_mar21a\sketch_mar21a.ino: In function 'void setup()':

sketch_mar21a:78:20: error: no matching function for call to 'ESP32_FTPClient::WriteData(fs::File*)'

ftp.WriteData(&file);

                ^
In file included from C:\Users\rafa\Documents\Arduino\sketch_mar21a\sketch_mar21a.ino:4:0:

C:\Users\rafa\Documents\Arduino\libraries\esp32_ftpclient\src/ESP32_FTPClient.h:42:8: note: candidate: void ESP32_FTPClient::WriteData(unsigned char*, int)

void WriteData (unsigned char * data, int dataLength);

    ^
C:\Users\rafa\Documents\Arduino\libraries\esp32_ftpclient\src/ESP32_FTPClient.h:42:8: note: candidate expects 2 arguments, 1 provided

Se encontraron varias bibliotecas para "WiFi.h"
Usado: C:\Users\rafa\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
No usado: C:\Program
Se encontraron varias bibliotecas para "ESP32_FTPClient.h"
Usado: C:\Users\rafa\Documents\Arduino\libraries\esp32_ftpclient
Se encontraron varias bibliotecas para "FS.h"
Usado: C:\Users\rafa\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS
Se encontraron varias bibliotecas para "SPI.h"
Usado: C:\Users\rafa\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI
Se encontraron varias bibliotecas para "SD_MMC.h"
Usado: C:\Users\rafa\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD_MMC
Se encontraron varias bibliotecas para "SPIFFS.h"
Usado: C:\Users\rafa\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS
exit status 1
no matching function for call to 'ESP32_FTPClient::WriteData(fs::File*)'

Este informe podría contener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

ldab, 2020-03-22:
Please share your code.

Vicentetomas, 2020-03-22:
Thank you for your quick answer. It's the example
/******************************************************************************
ESP32-CAM remote image access via FTP. Take pictures with ESP32 and upload it via FTP making it accessible for the outisde network.
Leonardo Bispo
July - 2019
https://github.com/ldab/ESP32_FTPClient
//1 - Install SPIFF Library and choose 2MB minimum for the file example in the data directory
https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/
Documentation about mandatories libraries
https://github.com/espressif/arduino-esp32/blob/master/libraries/FS/src/FS.cpp
https://github.com/espressif/arduino-esp32/blob/master/libraries/FS/src/FS.h
https://github.com/ldab/ESP32_FTPClient/blob/master/src/ESP32_FTPClient.h
//2 - Save the *.ino example
//3 - Configure WIFI and Server FTP & Run it
Distributed as-is; no warranty is given.
******************************************************************************/
#include "Arduino.h"
#include <WiFi.h>
#include <WiFiClient.h>
#include <ESP32_FTPClient.h>
#include "octocat.h"
//Adding SPIFFS library
#include "SPIFFS.h"

//Update wifi parameters and ftp server parameters
#define WIFI_SSID ""
#define WIFI_PASS ""

char ftp_server[] = "";
char ftp_user[]   = "";
char ftp_pass[]   = "";

// you can pass a FTP timeout and debbug mode on the last 2 arguments
ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

void setup()
{
  Serial.begin( 115200 );

  WiFi.begin( WIFI_SSID, WIFI_PASS );

  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  //Open Fle from SPIFFS and get stream
  if(!SPIFFS.begin(true)){
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }

  File file = SPIFFS.open("/ESP32-WROOM-32_BANNER_1_0.jpg");
  if(!file){
    Serial.println("Failed to open file for reading");
    return;
  }

  ftp.OpenConnection();

  // Get directory content
  ftp.InitFile("Type A");
  String list[128];
  ftp.ContentList("", list);
  Serial.println("\nDirectory info: ");
  for(int i = 0; i < sizeof(list); i++)
  {
    if(list[i].length() > 0)
      Serial.println(list[i]);
    else
      break;
  }

  // Create the new file and send the image
  ftp.InitFile("Type I");
  ftp.NewFile("ESP32-WROOM.jpg");
  //Becarefull file reference
  ftp.WriteData(&file);
  ftp.CloseFile();

  //Close Stream from SPIFFS
  file.close();

  ftp.CloseConnection();
}

void loop()
{

}

ldab, 2020-03-22:
As you can see here https://github.com/ldab/ESP32_FTPClient/blob/8bba164143fb00f0bacfd93d0b143015775a2b7b/src/ESP32_FTPClient.h#L42 `WriteData` takes a char array and you're passing a SPIFFS file, therefore it will not work.

Check this issue discussed previously: https://github.com/ldab/ESP32_FTPClient/issues/10

closed 2020-03-22 by ldab