ldab/ESP32_FTPClient #14: Create folder if it doesn't exist.

Schaggo, 2020-04-27:
Hi, 

I am trying to check if a folder exists and if not create it. 
The folder structure  is "/images/[imageset]/monochrome",  "/images/[imageset]/color".  As in /images/1/monochrome, /images/2/monochrome, ...

When uploading initially via 
````
   String ftpprobestring = "/images/img" + String(counter) ;
   String ftpprobestring1 = ftpprobestring + "/col";
   String ftpprobestring2 = ftpprobestring + "/bw";
   ftp.OpenConnection();
   ftp.InitFile("TYPE A");
  
  ftp.MakeDir(ftpprobestring.c_str());
   ftp.MakeDir(ftpprobestring1.c_str());
   ftp.MakeDir(ftpprobestring2.c_str());
   ftp.CloseConnection();
````
every the program works like expected.
But if I want to create a folder that already exists I get a "550" for each attempt to create a folder - even if the following folders do not exist. For example the folder /1/ exists but the folders /monocrome/ and /color/ do not exist - I still get 3 answers "550". 

I tried to create a list and check if one of the strings in said list contains the respective expressions - no success. 

Is there a way to check if a folder exists and then create it?
Or use the server message like ftp.changeworkdir; if (server message == 550 ) makedir?

Cheers

ldab, 2020-04-27:
> I am trying to check if a folder exists and if not create it.

How?

> But if I want to create a folder that **already exists** I get a "550" for each attempt to create a folder - even if the following folders **do not exist**. For example the folder /1/ exists but the folders /monocrome/ and /color/ do not exist - I still get 3 answers "550".

The highlighted was  confusing, can you please elaborate?

> Is there a way to check if a folder exists and then create it?

Not as a FTP commands: https://en.wikipedia.org/wiki/List_of_FTP_commands
You will need to do it manually, by checking if the folder exists. You can use `ContentList(const char * dir, String * list)` in order to check if the folder exists.

Schaggo, 2020-04-27:
Hey thanks a lot for the response.
And also thanks A LOT for providing this library! 

>How? 

I first tried with a string list via ContentList  and a for loop to check if an entry contains the dynamic variable. I encountered a problem that the list got too big for the memory I have available. 

But I found your answer to https://github.com/ldab/ESP32_FTPClient/issues/11 
This made me understand the Getftpanswer function. 
So I added a an "Exists" function that is basically copy paste of ChangeWorkDir and a ExistStatus function that is a copy paste of GetftpAnswer, but sets an external flag "doesexist" instead of _isconnected. 
 
I'm pretty sure there is a more elegant way, but this seems to work in the situations described above and doesn't use much memory.

Thanks again.

ldab, 2020-04-27:
Sure that are better implementation, working with strings take a lot of memory, it would be better with char array, etc...

FTP by definition have some limitations :)

closed 2020-04-27 by ldab