ldab/ESP32_FTPClient #15: Issue when overwritting a file

benny007, 2020-06-06:
Hi,

Thanks for the work on this library.

I'm trying to use it to overwrite the same file again and again to a FTP server in the same stream (that is without closing the connection because it has to be very fast and closing the file, reopening it and closing and reopening the connection add a huge delay).

However the file on my server is not properly overwritten it just keeps on growing instead (the size of the file gets bigger and bigger)

Any thought if this expected or if there is something that can be done to overwrite the file without having to reopen a new data connection every time ?

Thanks.

ldab, 2020-06-06:
What code/function are you using?

benny007, 2020-06-06:
Something like this:

void setup(){
 ftp.OpenConnection();
 ftp.InitFile("Type I");
 ftp.ChangeWorkDir("/stream");
 ftp.NewFile("photo.jpg");  
}

void loop{
 stream()
}

void stream(void){
 cam.run();
 ftp.WriteData(cam.getfb(), cam.getSize());    
}

So it's actually taking the picture from the camera and uploading it to the FTP server, over and over again to stream it through a single .jpg file. But doing it this way result in having a growing file in my FTP server instead of overwritten it.  It's like there is nothing saying it's the end of the file.

And if I use the NewFile()/CloseFile() int the stream function this trigger a new connection every time hence producing a huge delay.

Cheers

ldab, 2020-06-06:
Such function is not natively on FTP https://en.m.wikipedia.org/wiki/List_of_FTP_commands

You may want to implement a routine to check if the file exists, delete it and write... However I don't know if it would work.

closed 2020-06-06 by ldab