ldab/ESP32_FTPClient #16: IP address of the response for PASV command should be referred

Qamada, 2020-07-28:
IP address of the response for PASV command should be referred like the below, because some of ftp servers set other IP addresses than itself.

`void ESP32_FTPClient::InitFile(const char* type){`
`  FTPdbgn("Send TYPE");`
`  if(!isConnected()) return;`
`  FTPdbgn(type);`
`  client.println(F(type));`
`  GetFTPAnswer();`
``
`  FTPdbgn("Send PASV");`
`  client.println(F("PASV"));`
`  GetFTPAnswer();`
``
`  char *tStr = strtok(outBuf, "(,");`
`  int array_pasv[6];`
`  for ( int i = 0; i < 6; i++) {`
`    tStr = strtok(NULL, "(,");`
`    if (tStr == NULL) {`
`      FTPdbgn(F("Bad PASV Answer"));`
`      CloseConnection();`
`      return;`
`    }`
`    array_pasv[i] = atoi(tStr);`
`  }`
`  unsigned int hiPort, loPort;`
`  hiPort = array_pasv[4] << 8;`
`  loPort = array_pasv[5] & 255;`
`  `
`  //Qamada added`
`  IPAddress pasvServer(array_pasv[0],array_pasv[1],array_pasv[2],array_pasv[3]);`
`  //Qamada added`

`  FTPdbg(F("Data port: "));`
`  hiPort = hiPort | loPort;`
`  FTPdbgn(hiPort);`
`  //Qamada modified`
`  //if (dclient.connect(serverAdress, hiPort, timeout)) {`
`  if (dclient.connect(pasvServer, hiPort, timeout)) {`
`    FTPdbgn(F("Data connection established"));`
`  }`
`}`

I test this code on ftp server of http://web.fc2.com/en/ in order to evaluate this modification.

ldab, 2020-07-28:
Hi,

Thanks for the heads up, I have tried the FC2 server as you pointed and the result is the same, no need for parsing the IP again:

```
Status:	Disconnected from server
Status:	Resolving address of ldab.web.fc2.com
Status:	Connecting to 104.244.99.22:21...
Status:	Connection established, waiting for welcome message...
Response:	220 web.fc2.com FTP Server
Command:	AUTH TLS
Response:	500 AUTH not understood
Command:	AUTH SSL
Response:	500 AUTH not understood
Status:	Insecure server, it does not support FTP over TLS.
Command:	USER ldab
Response:	331 Password required for ldab
Command:	PASS **********
Response:	230 User ldab logged in
Status:	Server does not support non-ASCII characters.
Status:	Logged in
Status:	Retrieving directory listing...
Command:	PWD
Response:	257 "/" is the current directory
Command:	TYPE I
Response:	200 Type set to I
Command:	PASV
Response:	227 Entering Passive Mode (104,244,99,22,243,129).
Command:	LIST
Response:	150 Opening BINARY mode data connection for file list
Response:	226 Transfer complete
Status:	Directory listing of "/" successful
```

closed 2020-07-28 by ldab

Qamada, 2020-07-28:
Hi ldab,

Thank you for your swift testing. I really appreciate that you released ESP32 ftp client.

I'm sorry but what you observed is NOT what always happens.

Time to time, fc2 ftp server set the same address for PASV IP address as you connect firstly. However, it would set other IP address than itself as I mentioned.

You can see the below case also.
Firstly, it connected to 104.244.99.132:21.
However, IP address of PASV response is 104.244.99.45:63669.

```
ホスト xxx.web.fc2.com (104.244.99.132:21) に接続しています.
接続しました.
>USER xxx
331 Password required for xxx
>PASS [xxxxxx]
230 User xxx logged in
>FEAT
211-Features:
 CLNT
 EPRT
 EPSV
 HOST
 LANG en-US*
 MDTM
 MFF modify;UNIX.group;UNIX.mode;
 MFMT
 MLST modify*;perm*;size*;type*;unique*;UNIX.group*;UNIX.groupname*;UNIX.mode*;UNIX.owner*;UNIX.ownername*;
 REST STREAM
 SITE MKDIR
 SITE RMDIR
 SITE SYMLINK
 SITE UTIME
 SIZE
 TVFS
 UTF8
211 End
>OPTS UTF8 ON
200 UTF8 set to on
>>CD C:\Users\xxx
>CWD /livecamera
250 CWD command successful
>XPWD
257 "/livecamera" is the current directory
>TYPE A
200 Type set to A
>PASV
227 Entering Passive Mode (104,244,99,45,248,181).
ダウンロードのためにホスト 104.244.99.45 (104.244.99.45:63669) に接続しています.
接続しました.
>MLSD
150 Opening ASCII mode data connection for MLSD
226 Transfer complete
ファイル一覧の取得は正常終了しました. (689 Bytes)
```

I replaced user ID string for security purpose.

Basically, I believe that it's much better for us to follow the below document.
ftp://ftp.rfc-editor.org/in-notes/rfc959.txt
However, if you are good enough to follow just your ftp server to release this useful library, that's fine. In that case, I'd like to release my version of ESP32 FTP client class.

Thanks,