ldab/ESP32_FTPClient #17: FTP port hardcoded?

mikestarkenburg, 2020-07-31:
Hi there;  Thanks for writing this guys, it is coming in really handy.  

Question:  Is there any reason that the FTP port is hardcoded to 21?  What do you think about taking that as an additional variable (and maybe with a default of 21?) 

A couple of FTP servers we're accessing use alternate ports (I suppose for "security" but...), I got around it by changing the hardcoded value (in ESP32_FTPClient.cpp at line 132).  I am sure that will bite me somehow in the future, but it worked!  

Just thinking I might not be the only guy with this need, thought I would ask.

M

ldab, 2020-08-03:
No reason for been hardcoded, added: 29bbefe3ff9a6f6499b86809da362aadac4a1418

closed 2020-08-03 by ldab

TranquilloCosta, 2021-10-15:
> No reason for been hardcoded, added: [29bbefe](https://github.com/ldab/ESP32_FTPClient/commit/29bbefe3ff9a6f6499b86809da362aadac4a1418)

@mikestarkenburg  mikestarkenburg commented on 31 Jul 2020 that the port number is hard coded = 21 in your library. He wanted to suggest to make it free to choose when initializing the client.

There is a duplicate of the initialisation definition where the second overwrites the first. They differ in `port = _port;` and ` port = 21;` Therefore, the port can only be changed within the source code of the library:

`ESP32_FTPClient::ESP32_FTPClient(char* _serverAdress, uint16_t _port, char* _userName, char* _passWord, uint16_t _timeout, uint8_t _verbose){
  userName = _userName;
  passWord = _passWord;
  serverAdress = _serverAdress;
  port = _port;
  timeout = _timeout;
  verbose = _verbose;
}`

`ESP32_FTPClient::ESP32_FTPClient(char* _serverAdress, char* _userName, char* _passWord, uint16_t _timeout, uint8_t _verbose){
  userName = _userName;
  passWord = _passWord;
  serverAdress = _serverAdress;
  port = 21;
  timeout = _timeout;
  verbose = _verbose;
}`

ldab, 2021-10-15:
Thanks for you comment but it’s incorrect, please refer to the following link which explain the implementation https://en.cppreference.com/w/cpp/language/default_arguments