ldab/ESP32_FTPClient #18: FTP error : Offline

daffaryn, 2020-08-15:
Hello, is any condition that make ftp show "FTP error : Offline" when uploading file? i had a problem with` ftp.OpenConnection();` sometimes when try to uploading file. Thanks for your answer

ldab, 2020-08-15:
Please provide more details about your code and potentially the serial monitor output

daffaryn, 2020-08-16:
The code :
`ftp.OpenConnection();

  //Create a file and write the image data to it;
  ftp.InitFile("Type I");
  ftp.ChangeWorkDir("assets/img/presensi"); // change it to reflect your directory
  const char *f_name = pic_name.c_str();
  ftp.NewFile( f_name );
  ftp.WriteData(fb->buf, fb->len);
  ftp.CloseFile();

  DBG("upload pict done");

  // Breath, withouth delay URL failed to update.
  delay(100);
  esp_camera_fb_return(fb);`

result on serial monitor:
`Uploading via FTP
FTP error : Offline
FTP error : Offline
upload pict done`

ldab, 2020-08-17:
Where does ```Uploading via FTP``` comes from?

enable verbose = 2 at the declaration for example ```ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);```

The "Offline" error is probably coming from the following line which would indicate either your server is offline or incorrect attributes:

https://github.com/ldab/ESP32_FTPClient/blob/29bbefe3ff9a6f6499b86809da362aadac4a1418/src/ESP32_FTPClient.cpp#L147

closed 2020-08-17 by ldab

KennerCost, 2020-09-24:
didn't working. FTP error : Offline

or  ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

..rebooting the esp32 many times

kinba3000, 2021-01-14:
Hey, i have the same issue. But my FTP Log says, that the ESP32 opend a session. But the same error is displayed.

/*
 * ESP32cam_ftp
 * Blog: http://www.gsampallo.com/blog/?p=686
 * Twitter: @gsampallo.com
 */
#include "esp_camera.h"
#include "soc/soc.h"           // Disable brownour problems
#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include "driver/rtc_io.h"
#include <WiFi.h>
#include <WiFiClient.h>   
#include "ESP32_FTPClient.h"

#include <NTPClient.h> //For request date and time
#include <WiFiUdp.h>
#include "time.h"

char* ftp_server = "192.168.0.10";
char* ftp_user = "test";
char* ftp_pass = "test";
char* ftp_path = "/timelapes/";

const char* WIFI_SSID = "";
const char* WIFI_PASS = "";

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org", (-3600*3), 60000);

ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

// Pin definition for CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

camera_config_t config;

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
 
  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASS);
  
  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.println("Connecting to WiFi..");
      
  }
  Serial.println("IP address: ");
      
  Serial.println(WiFi.localIP());

  initCamera();

  timeClient.begin();
  timeClient.update();

  Serial.println(timeClient.getFormattedTime());

  ftp.OpenConnection();
}

void loop() {
  timeClient.update();
  takePhoto();
  delay(1000);
}

void initCamera() {
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG; 
  
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;//FRAMESIZE_UXGA; // FRAMESIZE_ + QVGA|CIF|VGA|SVGA|XGA|SXGA|UXGA
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }  
  // Init Camera
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }  
}

void takePhoto() {

      
  camera_fb_t * fb = NULL;
  
  
  // Take Picture with Camera
  fb = esp_camera_fb_get();  
  if(!fb) {
    Serial.println("Camera capture failed");
    return;
  }

  /*
   * Upload to ftp server
   */
  

 /*ftp.ChangeWorkDir(ftp_path); */
  ftp.InitFile("Type I");

  String nombreArchivo = timeClient.getFormattedTime()+".jpg"; // AAAAMMDD_HHMMSS.jpg
  Serial.println("Subiendo "+nombreArchivo);
  int str_len = nombreArchivo.length() + 1; 
 
  char char_array[str_len];
  nombreArchivo.toCharArray(char_array, str_len);
  
  ftp.NewFile(char_array);
  
  ftp.WriteData( fb->buf, fb->len );
  Serial.println("test");
  ftp.CloseFile();
  
  /*
   * Free
   */
  esp_camera_fb_return(fb); 
}

Serial Monitor:
20:22:06.974 -> Connecting Wifi...
20:22:07.486 -> Connecting to WiFi..
20:22:07.486 -> IP address: 
20:22:07.486 -> 192.168.0.62
20:22:22.917 -> 03:28:33
20:22:22.917 -> Connecting to: 192.168.0.10
20:22:22.917 -> Command connected
20:22:22.952 -> Send USER
20:22:27.956 -> FTP error: Offline
20:22:27.956 -> Send PASSWORD
20:22:32.958 -> FTP error: Offline
20:22:32.958 -> Send SYST
20:22:37.980 -> FTP error: Offline
20:22:53.064 -> Send TYPE
20:22:53.064 -> FTP error: Offline
20:22:53.064 -> Subiendo 03:29:03.jpg
20:22:53.098 -> Send STOR
20:22:53.098 -> FTP error: Offline
20:22:53.098 -> Writing
20:22:53.098 -> FTP error: Offline
20:22:53.098 -> test
20:22:53.098 -> Close File

FTPserver log:
2021-01-14 20:22:22,576 server proftpd[20092] localhost (192.168.0.62[192.168.0.62]): FTP session opened.
2021-01-14 20:23:22,811 server proftpd[20092] localhost (192.168.0.62[192.168.0.62]): USER test: Login successful.

I don't know wehre the issue is. There is not much about on google.
Eventually you see something that i made wrong. 

FTP Server:
ProFTP

ldab, 2021-01-15:
Please open your own issue.

Your code returns "Offline" here: https://github.com/ldab/ESP32_FTPClient/blob/29bbefe3ff9a6f6499b86809da362aadac4a1418/src/ESP32_FTPClient.cpp#L70

Which indicates the client is not available, did you try with a different FTP server?

kinba3000, 2021-01-15:
No, wich FTP programm would you recomend for Debian? I realiesed that i have some issues with the server, but it worked with filezilla. I will try it with a diffrent FTP server fore sure.

ldab, 2021-01-15:
I use an online server, for example https://infinityfree.net/ or https://www.000webhost.com/