ldab/ESP32_FTPClient #21: FTP error: Offline

kinba3000, 2021-01-16:
I opend a new issue like you told me:
Hey, i have the same issue. But my FTP Log says, that the ESP32 opend a session. But the same error is displayed.

/*
 * ESP32cam_ftp
 * Blog: http://www.gsampallo.com/blog/?p=686
 * Twitter: @gsampallo.com
 */
#include "esp_camera.h"
#include "soc/soc.h"           // Disable brownour problems
#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include "driver/rtc_io.h"
#include <WiFi.h>
#include <WiFiClient.h>   
#include "ESP32_FTPClient.h"

#include <NTPClient.h> //For request date and time
#include <WiFiUdp.h>
#include "time.h"

char* ftp_server = "192.168.0.10";
char* ftp_user = "test";
char* ftp_pass = "test";
char* ftp_path = "/timelapes/";

const char* WIFI_SSID = "";
const char* WIFI_PASS = "";

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org", (-3600*3), 60000);

ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

// Pin definition for CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

camera_config_t config;

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
 
  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASS);
  
  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.println("Connecting to WiFi..");
      
  }
  Serial.println("IP address: ");
      
  Serial.println(WiFi.localIP());

  initCamera();

  timeClient.begin();
  timeClient.update();

  Serial.println(timeClient.getFormattedTime());

  ftp.OpenConnection();
}

void loop() {
  timeClient.update();
  takePhoto();
  delay(1000);
}

void initCamera() {
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG; 
  
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;//FRAMESIZE_UXGA; // FRAMESIZE_ + QVGA|CIF|VGA|SVGA|XGA|SXGA|UXGA
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }  
  // Init Camera
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }  
}

void takePhoto() {

      
  camera_fb_t * fb = NULL;
  
  
  // Take Picture with Camera
  fb = esp_camera_fb_get();  
  if(!fb) {
    Serial.println("Camera capture failed");
    return;
  }

  /*
   * Upload to ftp server
   */
  

 /*ftp.ChangeWorkDir(ftp_path); */
  ftp.InitFile("Type I");

  String nombreArchivo = timeClient.getFormattedTime()+".jpg"; // AAAAMMDD_HHMMSS.jpg
  Serial.println("Subiendo "+nombreArchivo);
  int str_len = nombreArchivo.length() + 1; 
 
  char char_array[str_len];
  nombreArchivo.toCharArray(char_array, str_len);
  
  ftp.NewFile(char_array);
  
  ftp.WriteData( fb->buf, fb->len );
  Serial.println("test");
  ftp.CloseFile();
  
  /*
   * Free
   */
  esp_camera_fb_return(fb); 
}

Serial Monitor:
20:22:06.974 -> Connecting Wifi...
20:22:07.486 -> Connecting to WiFi..
20:22:07.486 -> IP address: 
20:22:07.486 -> 192.168.0.62
20:22:22.917 -> 03:28:33
20:22:22.917 -> Connecting to: 192.168.0.10
20:22:22.917 -> Command connected
20:22:22.952 -> Send USER
20:22:27.956 -> FTP error: Offline
20:22:27.956 -> Send PASSWORD
20:22:32.958 -> FTP error: Offline
20:22:32.958 -> Send SYST
20:22:37.980 -> FTP error: Offline
20:22:53.064 -> Send TYPE
20:22:53.064 -> FTP error: Offline
20:22:53.064 -> Subiendo 03:29:03.jpg
20:22:53.098 -> Send STOR
20:22:53.098 -> FTP error: Offline
20:22:53.098 -> Writing
20:22:53.098 -> FTP error: Offline
20:22:53.098 -> test
20:22:53.098 -> Close File

FTPserver log:
2021-01-14 20:22:22,576 server proftpd[20092] localhost (192.168.0.62[192.168.0.62]): FTP session opened.
2021-01-14 20:23:22,811 server proftpd[20092] localhost (192.168.0.62[192.168.0.62]): USER test: Login successful.

I don't know wehre the issue is. There is not much about on google.
Eventually you see something that i made wrong.

FTP Server:
ProFTP

ldab, 2021-01-16:
Have you tried on different servers as I recommended? It sounds like the client was not available.

kinba3000, 2021-01-16:
I tried a diffrent server now. I used 000webhost.com. But it still gives me the same error. Do you know how i will get some more information out of the error? I even tried diffrent libaries versions.

ldab, 2021-01-16:
I tried just now with the example provided on this library + 000webhost.com and it works just fine. Maybe the problem is with your ISP?

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting Wifi...
....
IP address: 192.168.0.29
Connecting to: files.000webhost.com
Command connected
Send USER
Send PASSWORD
Send SYST
Send TYPE
Type A
Send PASV
Data port: 52285
Data connection established
Send CWD
Send MLSD
Result start
Result: 150 Connecting to port 31886
226-Options: -a -l
226 2 matches total
Result end

Directory info:
type=cdir;sizd=6;modify=20190811114938;UNIX.mode=0755;UNIX.uid=10183013;UNIX.gid=10183013;unique=804g6fc6ce21; .
type=pdir;sizd=4096;modify=20190712101201;UNIX.mode=0755;UNIX.uid=10183013;UNIX.gid=10183013;unique=804g6fc69002; ..
Send TYPE
Type A
Send PASV
Data port: 52183
Data connection established
Send MKD
Send CWD
Send TYPE
Type I
Send PASV
Data port: 30966
Data connection established
Send STOR
Writing
Close File
Send TYPE
Type A
Send PASV
Data port: 40375
Data connection established
Send STOR
Write File
Close File
Connection closed
```

kinba3000, 2021-01-16:
I will try it real quick

kinba3000, 2021-01-16:
I uploaded your example. Same error with 000webhost.com. I now even tried a diffrent esp32 cam. Wich model are you using? Do you insert a SD card?

ldab, 2021-01-16:
Please share the results from your test. I tried with a ESP32 pico.

kinba3000, 2021-01-16:
`Connecting Wifi...
22:43:21.904 -> .
22:43:21.904 -> IP address: 192.168.0.73
22:43:21.904 -> Connecting to: files.000webhost.com
22:43:30.902 -> FTP error: Offline
22:43:30.902 -> Send USER
22:43:35.911 -> FTP error: Offline
22:43:35.911 -> Send PASSWORD
22:43:40.912 -> FTP error: Offline
22:43:40.912 -> Send SYST
22:43:45.888 -> FTP error: Offline
22:43:45.888 -> Send TYPE
22:43:45.888 -> FTP error: Offline
22:43:45.958 -> Send CWD
22:43:45.958 -> FTP error: Offline
22:43:45.958 -> Send MLSD
22:43:45.958 -> FTP error: Offline`

I use a ESP32-cam AI Thinker

ldab, 2021-01-16:
Also share the code used on the test

kinba3000, 2021-01-16:
`/******************************************************************************
ESP32-CAM remote image access via FTP. Take pictures with ESP32 and upload it via FTP making it accessible for the outisde network. 
Leonardo Bispo
July - 2019
https://github.com/ldab/ESP32_FTPClient
Distributed as-is; no warranty is given.
******************************************************************************/
#include "Arduino.h"
#include <WiFi.h>
#include <WiFiClient.h> 
#include <ESP32_FTPClient.h>
#include "octocat.h"

#define WIFI_SSID "Hobi 2.4Ghz"
#define WIFI_PASS "password"

char ftp_server[] = "files.000webhost.com";
char ftp_user[]   = "fam-holinger";
char ftp_pass[]   = "password";

// you can pass a FTP timeout and debbug mode on the last 2 arguments
ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

void setup()
{
  Serial.begin( 115200 );

  WiFi.begin( WIFI_SSID, WIFI_PASS );
  
  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  ftp.OpenConnection();

  // Get directory content
  ftp.InitFile("Type A");
  String list[128];
  ftp.ChangeWorkDir("/public_html/zyro/gallery_gen/");
  ftp.ContentList("", list);
  Serial.println("\nDirectory info: ");
  for(int i = 0; i < sizeof(list); i++)
  {
    if(list[i].length() > 0)
      Serial.println(list[i]);
    else
      break;
  }

  // Make a new directory
  ftp.InitFile("Type A");
  ftp.MakeDir("my_new_dir");

  // Create the new file and send the image
  ftp.ChangeWorkDir("my_new_dir");
  ftp.InitFile("Type I");
  ftp.NewFile("octocat.jpg");
  ftp.WriteData( octocat_pic, sizeof(octocat_pic) );
  ftp.CloseFile();

  // Create the file new and write a string into it
  ftp.InitFile("Type A");
  ftp.NewFile("hello_world.txt");
  ftp.Write("Hello World");
  ftp.CloseFile();

  ftp.CloseConnection();
}

void loop()
{

}`

ldab, 2021-01-16:
My guess is that your router or internet provider blocks port 21. You can try to use your smartphone as a hotspot or other network, or other port and see if it works.

kinba3000, 2021-01-16:
No, my server stands at home. Port 21 is forwarded and i even tried connecting via LAN. There souldn't be any issues with my network. But i will try it real quick with my smartphone.

kinba3000, 2021-01-16:
sorry, can't test it with a diffrent network at the moment. Esp won't connect to my iPhone. But i am sure the network or the provider dosen't make any issues. With FileZilla it works fine to connect to 000webhost.com. Coneccting via CMD works to.

ldab, 2021-01-16:
I will close the issue as there's not enough evidence showing there is an issue with the library. I can reopen once the test was done on a different network.

closed 2021-01-16 by ldab

kinba3000, 2021-01-16:
okay, i will try it soon on a diffrent network. Thanks for you work and help!

kinba3000, 2021-01-16:
You were right. Diffrent network/ provider and it works. Not with my own ftp server, but with 000webhostng.com. There is surely no issue with your libary. Thanks for the greate work and the advices. I will now search the problem on my network.

ldab, 2021-01-16:
Try enabling the core and other debug to see if it shines a light. If using Arduino IDE I believe it's under tools->coredebug or something like that

Grandoni98, 2023-03-24:
> You were right. Diffrent network/ provider and it works. Not with my own ftp server, but with 000webhostng.com. There is surely no issue with your libary. Thanks for the greate work and the advices. I will now search the problem on my network.
> 
> Edit: I had my own dns settings in router settings. Ftp won't handel this right. The issue with my own FTP server was: DelayEngine was on and the server waitet some moments befor answering. In this time, the esp run in a timeout. After turning this off, all worked great.

Hello good evening. You know that I am testing the same code with an FTP server that I have created with Windows 10. I get the same error that you had but I have not been able to solve it yet. Could you explain to me how I deactivate the DelayEngine option of my server please.
Beforehand thank you very much.

kinba3000, 2023-03-24:
Hey, check your dns settings. I had pihole installed on my linux server. This made some issues wit the libarry if i remember correctly. Like shown above, try a diffrent FTP server outside your network. Hope its helps.