ldab/ESP32_FTPClient #23: ContentList(const char * dir, String * list); causes the download of the entire file

pcbtborges, 2021-02-09:
I am looking to confirm if the upload of a file have completed successfully and for that, after uploading the file I used the above command expecting a list of all files on the indicated directory (dir) but it takes a long time and brings the entire content of the directory.
My objective is to compare if the name and size of the file I transferred matches any file on the list so I can proceed and delete the file from local SPIFFS partition.

My command (taken from one of the examples, goes like this:
const char * dir = "/"   and 
String       list[128];

ContentList(const char * dir, String * list);

ldab, 2021-02-09:
This is not an issue with the library, closing

closed 2021-02-09 by ldab

pcbtborges, 2021-02-09:
Do You mean it is a problem with the FTP Server?
In case it is, is there a way to select a FTP server that works with the library?
Thanks

ldab, 2021-02-09:
No, what I meant is that what you wrote describes a misuse or misunderstanding of the commands, and not an issue.

pcbtborges, 2021-02-09:
OK, sorry, I tried to find some documentation but failed. Can you provide some source where I can learn the proper use for the methods provided by the library?
Thanks

ldab, 2021-02-09:
https://en.wikipedia.org/wiki/List_of_FTP_commands