ldab/ESP32_FTPClient #26: Unexpected FTP error:offline

uliatgithub, 2021-04-17:
Hi,,
In a little project I want to send a photo form an ESP32cam via FTP to a webserver. The server is my webpagehoster, so not a Filezilla Server within my own LAN domain.
To check out whether the FTP transfer from my ESP to the server works I used your upload code example.

Findings:
1. Accessing the FTP server works fine, Username and Password are accepted.

`
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Connecting Wifi...
.....
IP address: 192.168.2.173
Connecting to: 'myserver'.'myWebhoster'.com (I anonymized the server name)
Command connected
Send USER
Send PASSWORD
Send SYST
Send TYPE
Type A
Send PASV
Data port: 53624
Data connection established
`

2. Changing into a (beforehand with Filezilla defined) working Directory sees to work fine as well, even though there are not only the files listed, but lots of additional information. However, the (beforehand  with filezilla uploaded) file Termini.jpg was found.
`
Send CWD
Send MLSD
Result start
Result: 150 Opening ASCII mode data connection for MLSD
Result end
Directory info:
modify=20210417101532;perm=flcdmpe;type=cdir;unique=FD00U28833ED;UNIX.group=1004;UNIX.groupname=psacln;UNIX.mode=0755;UNIX.owner=10047;UNIX.ownername=web553f4; .
modify=20210417101137;perm=flcdmpe;type=pdir;unique=FD00U2689C0E;UNIX.group=1004;UNIX.groupname=psacln;UNIX.mode=0755;UNIX.owner=10047;UNIX.ownername=web553f4; ..
modify=20210417101533;perm=adfrw;size=23993;type=file;unique=FD00U287C1A8;UNIX.group=1004;UNIX.groupname=psacln;UNIX.mode=0644;UNIX.owner=xxxx;UNIX.ownername=myUsername; Termini.jpg`

3. Next command in the code are:
`// Make a new directory
  ftp.InitFile("Type A");
  ftp.MakeDir("my_new_dir");`
  
  The Serial Monitor shows from now on an Offline Error 
 `
Send TYPE
Type A
FTP error: Offline
Send PASV
FTP error: Offline
Bad PASV Answer
Connection closed
Send MKD
FTP error: Offline
Send CWD
FTP error: Offline
Send TYPE
FTP error: Offline
Send STOR
FTP error: Offline
Writing
FTP error: Offline
Close File
Send TYPE
FTP error: Offline
Send STOR
FTP error: Offline
Write File
FTP error: Offline
Close File
Connection closed
`
Any idea the Offline Error occurs?  
Anything I can check? 

Thanks!

ldab, 2021-04-17:
Please share your full code and serial monitor output

uliatgithub, 2021-04-17:
Here it is.
The code (99% your original example code):
/******************************************************************************

ESP32-CAM remote image access via FTP. Take pictures with ESP32 and upload it via FTP making it accessible for the outisde network. 
Leonardo Bispo
July - 2019
https://github.com/ldab/ESP32_FTPClient

Distributed as-is; no warranty is given.

******************************************************************************/
#include "Arduino.h"
#include <WiFi.h>
#include <WiFiClient.h> 
#include <ESP32_FTPClient.h>
#include "octocat.h"
#include "credentials.h"

// you can pass a FTP timeout and debbug mode on the last 2 arguments
ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

void setup()
{
  Serial.begin( 115200 );

  WiFi.begin( WIFI_SSID, WIFI_PASS );
  
  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  ftp.OpenConnection();

  // Get directory content
  ftp.InitFile("Type A");
  String list[128];
 ftp.ChangeWorkDir("/ESPFiles");
  ftp.ContentList("", list);
  Serial.println("\nDirectory info: ");
  for(int i = 0; i < sizeof(list); i++)
  {
    if(list[i].length() > 0)
      Serial.println(list[i]);
    else
      break;
  }

  // Make a new directory
  ftp.InitFile("Type A");
  ftp.MakeDir("my_new_dir");

  // Create the new file and send the image
  ftp.ChangeWorkDir("my_new_dir");
  ftp.InitFile("Type I");
  ftp.NewFile("octocat.jpg");
  ftp.WriteData( octocat_pic, sizeof(octocat_pic) );
  ftp.CloseFile();

  // Create the file new and write a string into it
  ftp.InitFile("Type A");
  ftp.NewFile("hello_world.txt");
  ftp.Write("Hello World");
  ftp.CloseFile();

  ftp.CloseConnection();
}

void loop()
{

}

------------------------------------------------------------------------
The Output of SM:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
�Connecting Wifi...
.....
IP address: 192.168.2.173
Connecting to: myserver.myhoster.com        <- anonymitzed
Command connected
Send USER
Send PASSWORD
Send SYST
Send TYPE
Type A
Send PASV
Data port: 63898
Data connection established
Send CWD
Send MLSD
Result start
Result: 150 Opening ASCII mode data connection for MLSD
Result end

Directory info:
modify=20210417101532;perm=flcdmpe;type=cdir;unique=FD00U28833ED;UNIX.group=1004;UNIX.groupname=psacln;UNIX.mode=0755;UNIX.owner=10047;UNIX.ownername=web553f4; .
modify=20210417101137;perm=flcdmpe;type=pdir;unique=FD00U2689C0E;UNIX.group=1004;UNIX.groupname=psacln;UNIX.mode=0755;UNIX.owner=10047;UNIX.ownername=myftpusername; ..
modify=20210417101533;perm=adfrw;size=23993;type=file;unique=FD00U287C1A8;UNIX.group=1004;UNIX.groupname=psacln;UNIX.mode=0644;UNIX.owner=10047;UNIX.ownername=myftpusername; Termini.jpg   <-- the testfile I uploaded with Filezilla
Send TYPE
Type A
FTP error: Offline
Send PASV
FTP error: Offline
Bad PASV Answer
Connection closed
Send MKD
FTP error: Offline
Send CWD
FTP error: Offline
Send TYPE
FTP error: Offline
Send STOR
FTP error: Offline
Writing
FTP error: Offline
Close File
Send TYPE
FTP error: Offline
Send STOR
FTP error: Offline
Write File
FTP error: Offline
Close File
Connection closed

-----------------------------------------------------------------
Credentials for the Wifi and FTP Server access are stored in my "credentials.h" header file, stored in the include directory (platformio).

P.S.: I didn't marked the code with the <> because at least in my view each line of code is appended to its predecessor. Unreadable...

ldab, 2021-04-17:
1 -  for code formatting one should use ```
2 - who hosts the webserver, goggle search for `webpagehoster` returns no relevant result?
3 - the error is thrown because the client is not available any longer, https://github.com/ldab/ESP32_FTPClient/blob/e5f89d4aa36c2a745ccf3d8d27bcad17410cb9af/src/ESP32_FTPClient.cpp#L70
4 - number 3 is also confirmed by "Connection closed"

My guess is that your server is closing the connection after few commands, some servers restrict the FTP access. As a solution you check if the connection is open and if closed, open again.

ldab, 2021-04-17:
1 -  for code formatting one should use ```
2 - who hosts the webserver, goggle search for `webpagehoster` returns no relevant result?
3 - the error is thrown because the client is not available any longer, https://github.com/ldab/ESP32_FTPClient/blob/e5f89d4aa36c2a745ccf3d8d27bcad17410cb9af/src/ESP32_FTPClient.cpp#L70

My guess is that your server is closing the connection after few commands, some servers restrict the FTP access. As a solution you check if the connection is open and if closed, open again.

closed 2021-04-17 by ldab

uliatgithub, 2021-04-18:
1 - thx, I great to know.
2 - my webhoster is [https://webhoster.ag/](url) . I didn't want to expose its name right away. 
3 - I isterted the openConnection method before the line  //Make a new directory:

```c++
  ftp.OpenConnection();     // reopens the connection
  
  // Make a new directory
  ftp.InitFile("Type A");
  ftp.MakeDir("my_new_dir");
```
With this modification the new directory "my_new_dir" was created successfully, and all files are uploaded to the new dirctory.

But, when the the connection is checked first if still open, the creation of the new directory and all other following commands fail.
```c++
  if(!ftp.isConnected()){
    ftp.OpenConnection();
  }
  // Make a new directory
  ftp.InitFile("Type A");
  ftp.MakeDir("my_new_dir");
  etc...
```
It seems with this modification the server has closed the connection, but the program doesn't know it.
That's a bit unfortunate, because how to predict when the connection will be closed and when it will stay on?  

When I use Filezilla the connection stays on quite long, only after a longer period of inactivity the connection is closed. 
When I send 'TYPE A' and 'MKD "my_new_dir"' as user defined commands (main menu->server->user defined commands) the server stays on.

Well, that's it.

ldab, 2021-04-18:
1 - You will notice that `ftp.OpenConnection();` is a bit dumb and only returns the value of the variable, it does not check if the TCP client is connected, to do so, you would need something like: `client.connected()` however `client` is a private variable, you can change it if you want.

2 - in order to find the reason for the disconnection you may want to add a `Serial.println(outBuf)` here: https://github.com/ldab/ESP32_FTPClient/blob/e5f89d4aa36c2a745ccf3d8d27bcad17410cb9af/src/ESP32_FTPClient.cpp#L87