ldab/ESP32_FTPClient #27: FTP error: 500 'MLSD': command not understood.

3Dietrich, 2021-09-07:
I tried to use ESP32_FTPClient to get ESP32 to send measurements via FileApend to a local file location via WiFi.
But it didn't work.. several reasons, this is the first:
Serial Monitor is repeating:```FTP error: 500 'MLSD': command not understood.```

Well, I think I found the reason, but I think that maybe others are also looking for this bug and then maybe a recommendation from the developer would be quite good.
So far my experience:
MLSD command is not 'original' FTP, so some devices probably don't offer it (my quite recent and popular Fritz!Box3490 for example). Without MLSD nothing works in the given example... what to do? 
The error occurs when calling
```
ftp.ContentList("", list);
```
and I found in the sourcecode that there is a workaround: instead of "ContentList" take "ContentListWithListCommand", so like this:
```
ftp.ContentListWithListCommand("", list);
```

If I am right, it would be great if this should be mentioned! If I am not right please give the correct anwer!

ldab, 2021-09-08:
MLSD is a FTP compliant command, you can find the full list of commands here: https://en.wikipedia.org/wiki/List_of_FTP_commands

Some FTP servers may not support `MLSD`, your server probably doesn't, therefore this is not a bug in this library, simply the command used is not supported by your server.

When you say you found `ContentListWithListCommand` in the source code, I'm unsure what you mean since this command is not present on this repository.

closed 2021-09-08 by ldab

3Dietrich, 2021-09-08:
yes, it's maybe not a bug for you, but if the [fritz.box](https://avm.de/produkte/fritzbox/), (at least in germany) a leading router company(!) doesn't support MLSD, then.. I don't know... at least your sketch doesn't work on this boxes. 
However using the same FTP Server with the apps 'Cyberduck' and 'Nimble commander' and 'VSCode' I don't have problems.. That's why I marked it as a bug.

3gyptian, 2022-06-01:
Thank you @3dietrich and @ldab!   But I agree on this comment .. as I had the same issue with my mac FTP server and fortunately found this closed issues for the solution.  

Perhaps add alternative list command commented out in the sample upload code?