ldab/ESP32_FTPClient #28: Bad PASV Answer

3Dietrich, 2021-09-08:
I use the example, filled with correct adresses and passwords, changed the ChangeWorkDir from ""/public_html/zyro/gallery_gen/" to  "/" and updated the 'ContentList' with the 'ContentListWithListCommand'..
I don't know.. it just doesn't work. I tried for hours.. 
here is the code without Wifi and ftp connections

```
ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

void setup()
{
  Serial.begin( 115200 );

  WiFi.begin( WIFI_SSID, WIFI_PASS );
  
  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  ftp.OpenConnection();

  // Get directory content
  ftp.InitFile("Type A");
  String list[128];
  ftp.ChangeWorkDir("/Dokumente");
//  ftp.ChangeWorkDir("/");
//  ftp.ContentList("", list);
  ftp.ContentListWithListCommand("", list);

  Serial.println("\nDirectory info: ");
  for(int i = 0; i < sizeof(list); i++)
  {
    if(list[i].length() > 0)
      Serial.println(list[i]);
    else
      break;
  }

  // Make a new directory
  ftp.InitFile("Type A");
  ftp.MakeDir("my_new_dir");

  // Create the new file and send the image
  ftp.ChangeWorkDir("my_new_dir");
  ftp.InitFile("Type I");
  ftp.NewFile("octocat.jpg");
  ftp.WriteData( octocat_pic, sizeof(octocat_pic) );
  ftp.CloseFile();

  // Create the file new and write a string into it
  ftp.InitFile("Type A");
  ftp.NewFile("hello_world.txt");
  ftp.Write("Hello World");
  ftp.CloseFile();

  ftp.CloseConnection();
}
void loop(){}
```

then i get this Serial Output:
```
Connecting Wifi...
.....
IP address: 192.168.0.119
Connecting to: 192.168.0.1
Command connected
Send USER
Send PASSWORD
Send SYST
Send TYPE
Type A
Send PASV
Data port: 43174
Data connection established
Send CWD
Send LIST
Result start
Result: 150 Opening ASCII mode data connection for '/bin/ls -lgA'.
Result end

Directory info: 
FRITZ-NAS.txt

Produkthandbuch.html

Send TYPE
Type A
Send PASV
Bad PASV Answer
Connection closed
Send MKD
FTP error: Offline
Send CWD
FTP error: Offline
Send TYPE
FTP error: Offline
Send STOR
FTP error: Offline
Writing
FTP error: Offline
Close File
Send TYPE
FTP error: Offline
Send STOR
FTP error: Offline
Write File
FTP error: Offline
Close File
Connection closed
```

I don't know why the second PASV is bad. And why is the whole ftp Offline right away - because the bad PASV? why? I am a bit desperate, there is no alternate Arduino-FTP but this one does not work at all for me (I tried 2 different boards).. Should I try something else? SSH?

ldab, 2021-09-08:
try to install a FTP Client like FileZilla, enabled the detailed log and inspect how your server behaves.

closed 2021-09-08 by ldab

3Dietrich, 2021-09-08:
If this Lib is for using ESP32 as Client, why I should use FileZilla? I used Finder, Cyberduck, NimbleCommander (on MacOS) and works ok, no errors using this clients. But I want to use **ESP32** this is why I tried the Arduino **ESP32_FTPClient** and I get no working connection when I use ESP32 with ESP32_FTPClient. At least it outputs some Errors - but this is not detailed enough? I really don't get your question.

ldab, 2021-09-08:
I didn't write a question, my comment earlier was a suggestion aiming to help you to understand how your server behaves. After knowing how your server behaves, you can work on implementing the solution for it.

If you search for the keyword "Offline" under issues, you will find few other users mentioned different problems and different solutions.

The library is intended to run on a less powerfull CPU then compared to your PC, it also have a lot less resources when features, it is not comparable.

3Dietrich, 2021-09-08:
in FileZilla I get on PASV this response:
```
2021-09-08 13:28:48 39435 1 Command: PASV
2021-09-08 13:28:48 39435 1 Response: 227 Entering Passive Mode (192,168,0,1,187,98)

```
What is bad in this answer?

leo-ublox, 2021-09-08:
It does not seem to be a bad answer, but in order to replicate the problem I would recommend manual sending all the commands included on the ESP32 log shared earlier.

ldab, 2021-09-08:
It does not seem to be a bad answer, but in order to replicate the problem I would recommend manual sending all the commands included on the ESP32 log shared earlier.

3Dietrich, 2021-09-08:
ok, I found 'one' solution: 
*Before* **every** InitFile or similiar I need to add
```ftp.OpenConnection();```
Now everything works! 
Ok, I need also to change the folder from root to /working folder ```ftp.ChangeWorkDir("/working folder");```  every time.
Strange. But it works!

hike6688, 2022-01-19:
Hi Idab,
nice library to save my bme280 logfiles from my esp32 device to my NAS system.
Had the same problem (Bad PASV Answer Connection closed .. ) as 3Dietrich when transfering the second log-file.
Solution is the same as 3Dietrich stated.
I followed the hint from 3Dietrich , now it works.

I keep the current workingDirPath in a static and call openConnectionAndChangeWorkingDir () right before each InitFile() call.
`
static String mWorkingDirPath = "";

void setWorkingDirPath(String pWorkingDirPath){
  Serial.println("my_ftpHelper::setWorkingDirPath: " + pWorkingDirPath);
  mWorkingDirPath = pWorkingDirPath;
}
String getWorkingDirPath(){
  return mWorkingDirPath;
}
String getBaseWorkingDirPath(){
  String result = "/full_path_of_base_directory_on_ftpserver/" ;
  return result;
}
boolean openConnectionAndChangeWorkingDir (){
  boolean result = false;
  ftp.OpenConnection();
  Serial.println ("my_ftphelper::openConnectionAndChangeWorkingDir-> try to change working dir to: " + mWorkingDirPath);
  ftp.ChangeWorkDir(getWorkingDirPath().c_str());
  if (ftp.isConnected()) {
    Serial.println("CWD set to done : " + mWorkingDirPath);
    result = true;
  } 
  return result;
}`