ldab/ESP32_FTPClient #29: FTP error: Offline

AndrewLivesChaos, 2022-02-28:
I read other issues about this error but I can't figure out what my problem. My own code didn't work so I create a new basic project (I'm using PlatformIO) from the example "upload_image". Then I followed [this](https://www.youtube.com/watch?v=GE8XSxRWfOI) tutorial from youtube that implements a local FTP server with Filezilla Server. My code now is 

```
/******************************************************************************
ESP32-CAM remote image access via FTP. Take pictures with ESP32 and upload it via FTP making it accessible for the outisde network. 
Leonardo Bispo
July - 2019
https://github.com/ldab/ESP32_FTPClient
Distributed as-is; no warranty is given.
******************************************************************************/
#include "Arduino.h"
#include <WiFi.h>
#include <WiFiClient.h> 
#include <ESP32_FTPClient.h>
#include "octocat.h"

#define WIFI_SSID "mywifissid"
#define WIFI_PASS "mywifipassword"

char ftp_server[] = "192.168.1.7";
char ftp_user[]   = "esp32";
char ftp_pass[]   = "myftppassword";

// you can pass a FTP timeout and debbug mode on the last 2 arguments
ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

void setup()
{
  Serial.begin( 115200 );

  WiFi.begin( WIFI_SSID, WIFI_PASS );
  
  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  ftp.OpenConnection();

  // Get directory content
  ftp.InitFile("Type A");
  String list[128];
  ftp.ChangeWorkDir("/public_html/zyro/gallery_gen/");
  ftp.ContentList("", list);
  Serial.println("\nDirectory info: ");
  for(int i = 0; i < sizeof(list); i++)
  {
    if(list[i].length() > 0)
      Serial.println(list[i]);
    else
      break;
  }

  // Make a new directory
  ftp.InitFile("Type A");
  ftp.MakeDir("my_new_dir");

  // Create the new file and send the image
  ftp.ChangeWorkDir("my_new_dir");
  ftp.InitFile("Type I");
  ftp.NewFile("octocat.jpg");
  ftp.WriteData( octocat_pic, sizeof(octocat_pic) );
  ftp.CloseFile();

  // Create the file new and write a string into it
  ftp.InitFile("Type A");
  ftp.NewFile("hello_world.txt");
  ftp.Write("Hello World");
  ftp.CloseFile();

  ftp.CloseConnection();
}

void loop()
{

}
```

The output of the code every time is

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0018,len:4

load:0x3fff001c,len:1044

load:0x40078000,len:10124

load:0x40080400,len:5828

entry 0x400806a8

Connecting Wifi...

........

IP address: 192.168.1.20

Connecting to: 192.168.1.7

FTP error: Offline
Send USER

FTP error: Offline
Send PASSWORD

FTP error: Offline
Send SYST

FTP error: Offline
Send TYPE

FTP error: Offline
Send CWD

FTP error: Offline
Send MLSD

FTP error: Offline

Directory info: 

Send TYPE

FTP error: Offline
Send MKD

FTP error: Offline
Send CWD

FTP error: Offline
Send TYPE

FTP error: Offline
Send STOR

FTP error: Offline
Writing

FTP error: Offline
Close File

Send TYPE

FTP error: Offline
Send STOR

FTP error: Offline
Write File

FTP error: Offline
Close File

Connection closed
```

I've tryed to connect with other FTP clients to my local server and it works. I read in the other similar issues that the problem may be due the server side configuration but I'm wondering why with other clients works.

ldab, 2022-02-28:
It fails to open the client connection with your server, are you sure the IP and port is correct? When the port is not defined, 21 is used.

AndrewLivesChaos, 2022-02-28:
I'm in local network. The IP of my server is correct (I can connect to it with other clients like Filezilla) and I can see the port 21 in the logs of success connections.

ldab, 2022-02-28:
Do you have the logs from the successful connection? Sharing may be helpful

AndrewLivesChaos, 2022-02-28:
I've resolved my issue. It was due to the firewall. My fault, sorry. But now I noticed that the FTP server needs to support the MLSD. With the Filezilla Server on Windows it's ok, but I'm trying to configure a FTP server on Raspberry and I have issues again.

Do you know a good FTP server for raspberry? I've tryed with proftpd but seems to have problem with some commands:

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0018,len:4

load:0x3fff001c,len:1044

load:0x40078000,len:10124

load:0x40080400,len:5828

entry 0x400806a8

ÿConnecting Wifi...

.......

IP address: 192.168.1.20

Connecting to: 192.168.1.111

Command connected

Send USER

Send PASSWORD

Send SYST

Send TYPE

Type A

Send PASV

Data port: 42469

Data connection established

Send MLSD

Result start

Result: 150 Opening ASCII mode data connection for MLSD

Result end

Directory info: 

 kakao

 ..

 .

Send TYPE

Type A

FTP error: Offline
Send PASV

FTP error: Offline
Bad PASV Answer

Connection closed

Send MKD

FTP error: Offline
Send CWD

FTP error: Offline
Send TYPE

FTP error: Offline
Send STOR

FTP error: Offline
Writing

FTP error: Offline
Close File

Send TYPE

FTP error: Offline
Send STOR

FTP error: Offline
Write File

FTP error: Offline
Close File

Connection closed
```

Thanks anyway.