ldab/ESP32_FTPClient #30: Unknown Commands

upiddy, 2022-04-12:
Hi,

I am having issues with error code 500 and the ftp server not understanding any commands. Does the server need any special programs installed to interpret this code? I have a pi 3b+ running a simple ftp server. The code I am running is the example upload code with just the wifi and server info changed. I cant get past the unknown commands to do things like change the directory. I should note that the connection to the server looks good its just when I start to issue commands for the server.

I understand that this may not be a bug but I would love any help you all are willing to give non the less. Thank you for your time.

[FTP_Code_Home.txt](https://github.com/ldab/ESP32_FTPClient/files/8475993/FTP_Code_Home.txt)

ldab, 2022-04-12:
What does the serial monitor prints?

upiddy, 2022-04-13:
Here is the serial output. It makes a data connection but when the code tried to execute the code I get this error. I should note that the change directory seems to work. If I input a non existent sub directory to change to it throws error 550: cant change directory.

[image]

ldab, 2022-04-13:
Use filezilla to connect to your raspberry Pi and check if commands are supported.

Google search indicates that it may be cause by your router firewall:
https://stackoverflow.com/questions/25210101/filezilla-stop-connecting-after-mlsd
https://forum.filezilla-project.org/viewtopic.php?t=13010

closed 2022-04-13 by ldab

upiddy, 2022-04-15:
Cool, Ill look into those links. Thank you for your help!

UNOSweeper, 2023-04-27:
Hi ldab., When I use your upload_image.ino then after the WIFI connection is established, in the Monitor I see:

15:22:43.814 -> Command connected
15:22:43.814 -> Send USER
15:22:43.905 -> Send PASSWORD
15:22:44.042 -> Send SYST
15:22:44.226 -> Send TYPE
15:22:44.226 -> Type A
15:22:44.317 -> Send PASV
15:22:44.505 -> Data port: 35304
15:22:44.552 -> Data connection established
15:22:44.552 -> Send CWD
15:22:44.740 -> Send MLSD
15:22:44.927 -> Result start
15:22:44.927 -> Result: 150 Opening ASCII mode data connection for MLSD
15:22:44.927 -> Result end
15:22:44.927 -> 
15:22:44.927 -> Directory info: 
15:22:44.927 -> modify=20230424124630;perm=flcdmpe;type=cdir;unique=FD00U20105314;UNIX.group=6103;UNIX.mode=0755;UNIX.owner=6103; .

15:22:44.927 -> modify=20230405071023;perm=flcdmpe;type=pdir;unique=FD00U2404EFE9;UNIX.group=6103;UNIX.mode=0755;UNIX.owner=6103; ..

15:22:44.972 -> Send TYPE
15:22:44.972 -> Type A
15:22:49.958 -> FTP error: Offline
15:22:49.958 -> Send PASV
15:22:54.966 -> FTP error: Offline
15:22:54.966 -> Bad PASV Answer

Which line in your ESP32_FTPClient.h is responsible for SEND PASV? 
Does this mean PASV mode? If so, can I also use Active mode?
I asked this because I still have the issue that after the dir listing is shown the FTP server then close apparently, because of the FTP error: Offline message. The execution only continues when I have inserted a new ftp.OpenConnection(). Then I have to insert a CWD as well etc.
My Service Provider suggested that the upload_image.ino maybe has to setup a Passive Mode. I think that's what's happening here.
My Service Provider told me that when using FileZilla the connection remains intact after doing a Listdir. I was already aware of that. I had the same experience using FileZilla.
Who is generating the message "Send MLSD"?
I read in your ESP32_FTPClient.h under "public" the function "void ContentList(const char * dir, String * list);"
I guess that this function is doing that, but where can I read where this function is described?
Who is generating the message "FTP error: Offline"? Is the FTP server sending that text?
When that stuff comes clear to me, then hopefully I can understand the conversation between the FTP server and the esp-module.
Sorry to ask these kinds of questions. It has to do with my lack of knowledge of progamming in C.

ldab, 2023-04-27:
*.h are header files, the functions are defined on *.c for example https://github.com/ldab/ESP32_FTPClient/blob/e5f89d4aa36c2a745ccf3d8d27bcad17410cb9af/src/ESP32_FTPClient.cpp#L193

UNOSweeper, 2023-05-09:
Hi ldab, Thanks for the link given in issue #37. After creating an account with 000webost.com, I couldn't access the file manager after clicking the file manager logo or through the https://panel.000webhost.com/dashboard/xxx/dashboard-pagina with Tools > file manager (where xxx is the name of my website). As a result, a redirect occurred that failed with the message: "This page is not reachable". (Temporarily switch of Firewall or Defender did not help. This is way too complicated for me to sort that out. I had another (mobile) network available and that worked there.
Incidentally enabling verbose = 2 and adding Serial.println(outBuf); at line 87 of the .cpp file is very usefull to monitor the Client-Server messages.
I continued to open your example "upload_image.ino" in Arduino IDE (v 1.8.19) and only changed the WIFI- andlogin details as well as  the correct folder path on line 47. Then compiled and uploaded to an AI Thinker ESP32-CAM module. No errors after execution. So if 000webhost.com's FTP server is used, everything goes according to plan. Unfortunately, this is not the case when I use my hosting provider's ftp server (as explained at the bottom of issue #30). In that case the execution goes well up to and including the ListDir. Then something goes wrong with FTP error: Offline after the ftp. InitFile("Type A") on line 59 is sent. I've read that other users experienced this error as well. 
After adding the commands ftp.OpenConnection() followed by ftp.ChangeWorkDir() then the execution is errorless. Sending only ftp.ClosFile() turned out to work as well with the advantage that directory change is not needed now.
So after adding only ftp. CloseFile() on the blank line 57 of your sample sketch, everything is now fine from start to finish. 
Not client.stop() but dclient.stop() as part of that command is apparently a solution. I can't explain why that is.
If desired, I can share the output of the Serial Monitor, when using both FTP servers. Should I use "Add code <Ctrl+e> " for that?