ldab/ESP32_FTPClient #32: Server does not seem to like change directory command

noweare1, 2023-03-03:
I am running vsftpd server on a linux desktop. I can connect to it and run commands successfully using a raspberry pi.
I am using the upload example program on an esp32. The esp32 connects but the server is choking on the change directory command.
Here is the output from the monitor:
IP address: 192.168.1.10
Connecting to: 192.168.1.8
Command connected
Send USER
Send PASSWORD
Send SYST
Send TYPE
Type A
Send PASV
Data port: 40313
Data connection established
FTP error: 550 Failed to change directory.

Send MLSD
FTP error: 550 Failed to change directory.

Directory info: 
Send TYPE
FTP error: 550 Failed to change directory.

Send STOR
FTP error: 550 Failed to change directory.

Writing
FTP error: 550 Failed to change directory.

Close File
Connection closed

Just wondering if you might have some ideas I can try.

Thank you.

ldab, 2023-03-03:
Are you sure the directory exists or the path is correct? Check for leading slashes and things like that

noweare1, 2023-03-03:
Yes, from my raspberry pi client, I can get a listing and download a test file from the /home/sammy/ftp/files directory.  I am sure it's me and not your code especially when there are 0 Issues (nice job).  From the pi, if i run the list command it shows I am at the {user}/ftp" directory. I will play around with it some more. I will figure it out eventually, thank you.

closed 2023-03-03 by ldab