ldab/ESP32_FTPClient #34: FTP error: offline

weijunying2019102969, 2023-04-14:
I read all those closed issue about offline, but I could not solve my problem yet.  my code is like this
`/******************************************************************************

ESP32-CAM remote image access via FTP. Take pictures with ESP32 and upload it via FTP making it accessible for the outisde network. 
Leonardo Bispo
July - 2019
https://github.com/ldab/ESP32_FTPClient

Distributed as-is; no warranty is given.

******************************************************************************/
#include "Arduino.h"
#include <WiFi.h>
#include <WiFiClient.h> 
#include <ESP32_FTPClient.h>
#include "octocat.h"

#define WIFI_SSID "IMTeam"
#define WIFI_PASS "IMTeamgogogo"

char ftp_server[] = "47.110.41.91";
char ftp_user[]   = "system";
char ftp_pass[]   = "OTA&mqtt1234.";

// you can pass a FTP timeout and debbug mode on the last 2 arguments
ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

void setup()
{
  Serial.begin( 115200 );

  WiFi.begin( WIFI_SSID, WIFI_PASS );
  
  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  ftp.OpenConnection();

  // Get directory content
  // ftp.InitFile("Type A");
  // String list[128];
  // ftp.ChangeWorkDir("/work/ftp/uploadFile");
  // ftp.ContentList("", list);
  // Serial.println("\nDirectory info: ");
  // for(int i = 0; i < sizeof(list); i++)
  // {
  //   if(list[i].length() > 0)
  //     Serial.println(list[i]);
  //   else
  //     break;
  // }

  // // Make a new directory
  // ftp.InitFile("Type A");
  // ftp.MakeDir("my_new_dir");

  // //Create the new file and send the image
  // ftp.ChangeWorkDir("my_new_dir");
  ftp.InitFile("Type I");
  ftp.ChangeWorkDir("/work/ftp/uploadFile/");
  ftp.NewFile("octocat.jpg");
  ftp.WriteData( octocat_pic, sizeof(octocat_pic) );
  ftp.CloseFile();

  // Create the file new and write a string into it
  ftp.InitFile("Type A");
  ftp.ChangeWorkDir("/work/ftp/uploadFile/");
  ftp.NewFile("hello_world.txt");
  ftp.Write("Hello World");
  ftp.CloseFile();

  ftp.CloseConnection();
}

void loop()
{

}`

and I got results like this :

`IP address: 192.168.10.108
Connecting to: 47.110.41.91
Command connected
Send USER
Send PASSWORD
Send SYST
Send TYPE
Type I
Send PASV
Data port: 34489
Send CWD
Send STOR
FTP error: Offline
Writing
FTP error: Offline
Close File
Send TYPE
FTP error: Offline
Send CWD
FTP error: Offline
Send STOR
FTP error: Offline
Write File
FTP error: Offline
Close File
Connection closed`

I can see those files that I uploaded just appear in my ftp server through FileZilla, but all of them are 0 byte size. 
So what is my peoblem with my ftp server or client

weijunying2019102969, 2023-04-14:
below are those files I uploaded. I listed it by cmd ftp client.
> 230 Login successful.
> ftp> cd work
> 250 Directory successfully changed.
> ftp> cd ftp
> 250 Directory successfully changed.
> ftp> cd uploadFile
> 250 Directory successfully changed.
> ftp> ls -l
> 200 PORT command successful. Consider using PASV.
> 150 Here comes the directory listing.
> -rwxrwxrwx    1 1000     1000            0 Apr 13 15:37 14.txt
> -rwxrwxrwx    1 1000     1000            0 Apr 13 15:39 15.txt
> -rwxrwxrwx    1 1000     1000            0 Apr 13 15:29 33.txt
> -rwxrwxrwx    1 1000     1000            0 Apr 13 18:15 hello_world.txt
> -rw-------    1 1000     1000            0 Apr 14 10:50 octocat.jpg
> 226 Directory send OK.

weijunying2019102969, 2023-04-14:
I locate the problem, it must be this function
```
void ESP32_FTPClient::NewFile (const char* fileName) {
  FTPdbgn("Send STOR");
  if(!isConnected()) return;
  client.print(F("STOR "));
  client.println(F(fileName));
  GetFTPAnswer();
}
```
I cannot understand what "client.println(F(fileName));" here mean ?

ldab, 2023-04-14:
> I locate the problem, it must be this function
>
> ```
> void ESP32_FTPClient::NewFile (const char* fileName) {
>   FTPdbgn("Send STOR");
>   if(!isConnected()) return;
>   client.print(F("STOR ")); [...]

This only set the filename on the server.

From what you shared you are receiving `FTP error: Offline` which have been reported few times before, have a look on the closed issue and you may find one solution that fixes it for you.

closed 2023-04-14 by ldab

ldab, 2023-04-21:
> Trying to upload a livecam picture with my ESP32-CAM board (so far not succeeded) using your ESP32-FTPClient header file, I want to understand what type of commands are enclosed in your ESP32-FTPClient.h or which commands are available to use. In your .ino file I read for example ftp.OpenConnection, ftp.Initfile, ftp.Makedir, ftp.WriteData and ftp.CloseFile and more. All these commands are sending something to the FTP server. Are there also ftp. commands to read what a FTP server is sending back (to catch an answer)? I can imagine that there are commands that provoke a response from the server. I read for example in your .cpp file a command like GetFTPAnswer(); Where can I find the available commands and what the syntaxes are, how to use them correctly by using them in a ino or cpp file? All these stuff is rather new for me and I am investigating why I sometimes have to add an extra ftp.OpenConnection in my ino sketch between a ListDIR and a command to create a new DIR at the FTP server, otherwise I see only the Dir listing and no new directory is created. Thank you in advance for your help.

You will find the answers for your questions by browsing on the repository files.

All commands sent to the server will have a server response, if you want to print them, you could add `ESP32_FTPClient( _serverAdress,  _userName,  , uint16_t _timeout,  _verbose)` have a look on other closed issue and you may find something that will help you.

Otherwise if you believe there is an issue with the code, please open a new issue and include a detail description of the issue and the necessary steps to reproduce it.