ldab/ESP32_FTPClient #35: FTP Offline

KvarTechnologies, 2023-04-15:
code:-
ESP32_FTPClient ftp1 (ftp_server_t,ftp_user_t,ftp_pass_t,3000,2);

void Send_FTP_file(fs::FS& fs, const char* path, ESP32_FTPClient ftpClient) {
    //Serial.printf("Reading file: %s\n", path);

    String fullPath = "/";
    fullPath.concat(path);
    Serial.print("File Path:");
    Serial.print(path);
    File file = SD.open(fullPath);
    if(!file){
        Serial.print("Failed to open file for reading");
        return;
    }
    
    ftpClient.InitFile("Type I");
    ftpClient.NewFile(path);
    char sz = file.size();
    int fileSize = file.size();
    Serial.print("File Size:");
    Serial.print(fileSize);   
    unsigned char * downloaded_file = (unsigned char *) malloc(fileSize);    
    size_t bytesRead = file.read(downloaded_file, fileSize); // read the file contents into the array   
    //const char* myString = reinterpret_cast<const char*>(downloaded_file);  // for converting unsigned char to const char*
    ftpClient.WriteData(downloaded_file, fileSize);   
    free(downloaded_file);
    ftpClient.CloseFile();
    file.close();

    
}

Error:
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
Send USER<CR><LF>
Send PASSWORD<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>
FTP error: Offline<LF>
Connecting to: 13.126.134.217<CR><LF>
Command connected<CR><LF>....

I remains in the condition unitl  FTP connected, Can u impalement no of tries in the FTPAnswer routine. To skip if tit exceeds no of tries . So that it can again try to reconnect when the function of send is called. Or is there any other way to tackle this issue

ldab, 2023-04-15:
I’ll not work on that, feel free to change as per your requirements.

closed 2023-04-15 by ldab

KvarTechnologies, 2023-04-17:
i am unable to figureout the exact issue with ftp : offline, it shows ftp offline and after few tries it connects. Again after few time , shows FTP : OFFLINE