ldab/ESP32_FTPClient #36: send MLSD -> FTP error: 500 unknown command.  vsfptd

grantiago, 2023-04-27:
I guess the answer is to install proftp. I'm using vsftpd on ubuntu. 
[stackoverflow desc of error.](https://stackoverflow.com/questions/49528887/ftplib-mlsd-command-gives-500-unknown-command)

```
Send CWD
12:30:23.963 -> Send MLSD
12:30:24.010 -> FTP error: 500 Unknown command.
```

[/ESP32_FTPClient.cpp](https://github.com/ldab/ESP32_FTPClient/tree/master)/[src](https://github.com/ldab/ESP32_FTPClient/tree/master/src)

line 257
 ```
FTPdbgn("Send MLSD");
  if(!isConnected()) return;
  client.print(F("MLSD"));
  client.println(F(dir));
  GetFTPAnswer(_resp);

```

ldab, 2023-04-27:
I’m unsure about what is the problem. But I believe you already sorted it out. Your server does not support MLSD

closed 2023-04-27 by ldab

grantiago, 2023-04-27:
Just for others that may encounter this error. To save them some time.