ldab/ESP32_FTPClient #38: myPhoto.png or myphoto.png in example download_file.ino?

UNOSweeper, 2023-05-09:
Hi ldab. I have executed your example download_file.ino using my 000webhost.com account.
On the Serial Monitor (output not shown here) I saw that myPhoto.png did not exist.
Reading your .ino file I made out at that I first had to create a picturefile with that name and had to upload that file into the working directory (public_html/xxx). After I had done so the Serial Monitor output (verbose=2 and Serial.println(outBuf) in the ESP32-FTPClient.cpp at line 87) was -> see List-1.txt.
[List-1.txt](https://github.com/ldab/ESP32_FTPClient/files/11435646/List-1.txt)

The result of this is that only helloworld.txt is written to the working directory /xxx 
and the Filesize (of myPhoto.png) is: 0. And MyNewDir is not created.
Then I renamed myPhoto.png in your download_file.ino into myphoto.png (all lowercase letters).
The output of the Serial Monitor  -> SeeList-2.txt.
[List-2.txt](https://github.com/ldab/ESP32_FTPClient/files/11435665/List-2.txt)

Finally I also renamed myPhoto.png in my working directory /xxx into myphoto.png (all lowercase letters).
The Serial Monitor output is in List-3.txt. [](url)
[List-3.txt](https://github.com/ldab/ESP32_FTPClient/files/11435674/List-3.txt)

I observe at line 77 of your .ino file list[i].toLowerCase(), but that is only the way how the files are shown in the list? Does it have anything to do with that?
Maybe I have done something wrong.
In general, I think it would be nice to add the Serial Monitor output to the examples as a reference comparison.

UNOSweeper, 2023-05-09:
.Sorry for using the three Serial monitor output text as a link. Maybe there is another way, which I did not find.

ldab, 2023-05-10:
If you believe there is an issue with the library, please share the code you used, the output of the terminal and the detailed steps on how to reproduce it.

UNOSweeper, 2023-05-10:
If I haven't written the issue in enough detail I apologize. 
I do not know where the issue is, but if you like to download myPhoto.png into a new map myNewDir, then that does not succeed with your example file ESP32_FTPClient/examples/download_file/download_file.ino
And according this sketch it is supposed that there must already exist a file myPhoto.png in the working directory on the server.

Below is described what I have done.
I downloaded your file ESP32_FTPClient/examples/download_file/download_file.ino.
I changed in that file the credentials at line 16...21
I changed the path on line 47 and 109 of your working directory into my working directory (public_html/xxx) at 000webhost.com .
I changed line 23 of the sketch into ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000,2); to have Verbose=2.
Then the sketch is:
`/******************************************************************************

ESP32-CAM remote image access via FTP. Take pictures with ESP32 and upload it via FTP making it accessible for the outisde network. 
Leonardo Bispo
July - 2019
https://github.com/ldab/ESP32_FTPClient

Distributed as-is; no warranty is given.

******************************************************************************/
#include "Arduino.h"
#include <WiFi.h>
#include <WiFiClient.h> 
#include <ESP32_FTPClient.h>

#define WIFI_SSID ""
#define WIFI_PASS ""

char ftp_server[] = "files.000webhost.com";
char ftp_user[]   = "";
char ftp_pass[]   = "";

ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000,2);

void setup()
{
  Serial.begin( 115200 );

  WiFi.begin( WIFI_SSID, WIFI_PASS );
  
  Serial.println("Connecting Wifi...");
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.print("\nMax Free Heap: ");
  Serial.println(ESP.getMaxAllocHeap());
  Serial.println("");

  ftp.OpenConnection();

  //Change directory
  ftp.ChangeWorkDir("/public_html/xxx");

  // Create a new file to use as the download example below:
  ftp.InitFile("Type A");
  ftp.NewFile("helloworld.txt");
  ftp.Write("Hi, I'm a new file");
  ftp.CloseFile();

  //Download the text file or read it
  String response = "";
  ftp.InitFile("Type A");
  ftp.DownloadString("helloworld.txt", response);
  Serial.println("The file content is: " + response);

[... 509 lines cut ...]
14:11:32.412 -> 226 0.000 seconds (measured here), 35.49 Mbytes per second
14:11:32.412 -> Result end
14:11:32.412 -> Send TYPE
14:11:32.412 -> Type I
14:11:33.200 -> Send PASV
14:11:33.847 -> Data port: 36088
14:11:33.985 -> Data connection established
14:11:33.985 -> Send MKD
14:11:34.354 -> Send CWD
14:11:34.769 -> Send STOR
14:11:35.137 -> Writing
14:11:35.368 -> Close File
14:11:35.553 -> Connection closed`

The File Manager of 000webhost.com now showed in my working directory /public_html/xxx :
 - myNewDir	
 - helloworld.txt
 - myphoto.png 

The File Manager of 000webhost.com showed in my working directory /public_html/xxx/myNewDir :
 - myphoto.png

So only when in the sketch (at line 62) the filename is written as myphoto.png and in the working directory /public_html/xxx 
the file myphoto.png is found then the sketch is generating the myNewDir map and is copying the file myphoto.png in it.
To make sure I tested this sketch also with an ESP32 Dev module. Same result.

ldab, 2023-05-10:
And what is the problem? on the first example you shared it prints `13:54:18.415 -> File size is: 0` therefore it is clear that the size is invalid and thus no file to download.

UNOSweeper, 2023-05-10:
The problem is that I experience that the filename to be downloaded with this sketch must have a lowercase name.
You are right, I agree, at that moment there was no myPhoto.png yet in my working directory.
After that moment I had created a .png file with that name you specified in the sketch as myPhoto.png and I have saved that file into my working directory.
File size 3.0 kB 
Running the sketch again, then there should be still no file (or invalid) in my working directory dat the Serial Monitor shows at 14:04:52.662 -> File size is: 0. 
However, I have showed you there exists a file with the name myPhoto.png of 3.0 kB in my working directory.

The run of 14:08:27 is only meant as an intermediate step to check what the result was when the file name in the sketch at line 62 is changed into myphoto.png and myPhoto.png in my working directory hasn't yet changed into myphoto.png.
The result of that step is 14:08:27.617 -> File size is: 3052 but the sketch can't open file myphoto.png.

Only when the specified filename in the sketch at line 62 consists of only a lowercase name (I changed it into myphoto.png) and there exists a file in my working directory with the same lowercase name (I changed that as well into myphoto.png ), only then the sketch generates the myNewDir subdirectory in my working directory and downloads that file in there (the run of 14:11:22.750), as the File Manager of 000webhost.com showed me.

ldab, 2023-05-10:
Aaa ok, but that is just a name mismatch that you can fix on your code. I don’t believe there is an issue with the library, so I’ll close this issue.

closed 2023-05-10 by ldab

UNOSweeper, 2023-05-11:
After playing around with the code of your published example ESP32_FTPClient/examples/download_file/download_file.ino I can tell you now that by simply deleting, removing or commenting   list[i].toLowerCase();   at line 77 of your code will solve the problem. Then the complete sketch will works as you intended. This is the case when using the FTP server of 000webhost.com.
Using the FTP server of my own ISP then I have to add ftp.CloseFile() before a next ftp.InitFile() is sent to let the sketch working correctly (same experience as with your ESP32_FTPClient/examples/upload_image/upload_image.ino. See issue #30).