ldab/ESP32_FTPClient #40: Parametric management of FTP parameters

biggiu, 2023-11-30:
Hello and congratulations for the good work done.
I tried the library and everything works great.
Now however, I would like to make the access parameters to the FTP area parametric.
I planned to read these parameters from SD-CARD and assign them to the variables:

char ftp_server[] = "";
char ftp_user[] = "";
char ftp_pass[] = "";

for example :

String FTP_SER;
String FTP_USR;
String FTP_PWD;

these variables are read from SD-CARD.

Then I try to do:
char ftp_server[] = FTP_SER;
char ftp_user[] = FTP_USR;
char ftp_pass[] = FTP_PWD;

but I can't get it to work no matter what I try.
It's true that I'm not very good with the Arduino language but it seems strange to me not to be able to manage something apparently so simple.

Would you be so kind as to give me a hand.
Thank you for any collaboration.

ldab, 2023-11-30:
You can't direct assign a `String` to a char array. I believe you need https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/tochararray/

closed 2023-11-30 by ldab

ldab, 2023-11-30:
On the code shared I do not see where it would print “ Connecting to:”

It is recommended to paste the full output of the terminal.

this is not related to this library so I’d suggest you post your question on arduino forum.

without spending more time on it, it seems like your char ftp_server etc is not big enough to contain the string.

biggiu, 2023-11-30:
Hello and thanks for the prompt response. I'll try to explain the problem differently:

I define the char variables:

   char ftp_server[] = "";
   char ftp_user[] = "";
   char ftp_pass[] = "";

I associate the values with String variables:

FTP_SER = "ftpserver";
FTP_USR = "ftpuser";
FTP_PWD = "ftppassword";

after using toCharArray I get:

ftp_server = ftpserver;
ftp_user = ftpuser;
ftp_pass = ftppassword

So far everything is fine and nothing strange.
Now, as soon as I start the function:

ESP32_FTPClient ftp (ftp_server,ftp_user,ftp_pass, 5000, 2);

The following message appears on the Arduino IDE serial monitor:

Connecting to: passwordftp

I put dummy values as FTP parameters so as not to display the real ones, but the result doesn't change.
The strange thing, which I don't think depends on Arduino, is this exchange of parameters. Which doesn't happen using your example.
If you spend some time testing the code you might find something that wasn't expected.

Thanks anyway for your great work.
Greetings.

ldab, 2023-11-30:
As I wrote earlier, you’re declaring the variables wrong.

try

char ftp_pass[64]…. Etc

biggiu, 2023-12-01:
Thanks for your patience, everything works perfectly now.
Good work