ldab/ESP32_FTPClient #46: Upload image from SD-CARD

biggiu, 2024-02-13:
Hello and sorry for the inconvenience.
I encountered a problem when I try to copy a jpg image from the SD-CARD to the FTP area.
The image is copied correctly but is unreadable.
I attach the part of code that performs this operation.
I would be grateful if you could help me solve the problem.
Best regards.
[UPLOAD.zip](https://github.com/ldab/ESP32_FTPClient/files/14257635/UPLOAD.zip)

ldab, 2024-02-13:
Paste your code here instead of a zip file

biggiu, 2024-02-14:
/************************************************/
    /* RICERCA FOTO VIAGGIO DA TRASFERIRE           */
    /************************************************/
    void CerFotVia(fs::FS &fs, const char * dirname, uint8_t levels)
    {
      Serial.printf("Listing directory: %s\n", dirname);

      File root = fs.open(dirname);
      if(!root)
      {
        Serial.println("Failed to open directory");
        return;
      }
      if(!root.isDirectory())
      {
        Serial.println("Not a directory");
        return;
      }

      File file = root.openNextFile();
      while(file)
      {
        if(file.isDirectory())
        {
          Serial.print("  DIR : ");
          Serial.println(file.name());
          if(levels)
          {
            CerFotVia(fs, file.path(), levels -1);
          }
        }
        else
        {
          Serial.print("  FILE: ");
          Serial.print(file.name());
          Serial.print("  SIZE: ");
          Serial.println(file.size());
          // Trasferimento Immagini del Viaggio
          NumDocCar +=1;
          //TraFotVia(file.name(),file.size());
          TraFotVia(file.path(),file.size());
          // Cancellazione Documento Trasferito
          DelDocVia(SD_MMC,file.path());
        }  
        file = root.openNextFile();
      }
    }

    /************************************************/
    /* CANCELLA DOCUMENTO VIAGGIO TRASFERITO SI FTP */
    /************************************************/
    void DelDocVia(fs::FS &fs, const char * path)
    {
      Serial.printf("Deleting file: %s\n", path);
      if(fs.remove(path))
      {
        Serial.println("File deleted");
      }
      else
      {
        Serial.println("Delete failed");
      }
    } 

    /************************************************/
    /* AVVIA TRASFERIMENTO IMMAGINI                 */
    /************************************************/
    void TraFotVia(String NomFot,int DimFot)
    {
      char Foto[40];
      boolean EsiConFtp = false;
      int NumTenTra = 0;

      // Forma Percorso Cartella Viaggio
      w0 = String(FTP_DIR) + String(DirViaTra) + "/";
      LW = w0.length()+1;
      w0.toCharArray(PatViaFtp, LW);

      Serial.printf("DirViaTra %s , PatViaFtp %s, DimFot %d NomFot , ",DirViaTra,PatViaFtp,DimFot);
      Serial.println(NomFot);

      ESP32_FTPClient ftp (FTP_SER,FTP_POR,FTP_USR,FTP_PWD, 5000, 2);

      // Apertura Immagine
      //w0 = String(DirViaTra) + "/" + NomFot;
      //Serial.println(w0);

      w0 = NomFot.substring(1);
      LW = w0.length()+1;
      w0.toCharArray(Foto, LW);

      while (EsiConFtp == false && NumTenTra < 3)
      {
      ftp.OpenConnection();
      if ( ftp.isConnected() )
        {        
          //ftp.ChangeWorkDir( PatViaFtp );        
          ftp.ChangeWorkDir( FTP_DIR );        
          ftp.InitFile("Type I");
          ftp.NewFile(Foto);
          ftp.WriteData( (unsigned char *)Foto,sizeof(Foto));
          ftp.CloseFile();
          ftp.CloseConnection();
          NumDocTra +=1;
          EsiConFtp = true;         
        }
        else
        {
          delay(500);
          NumTenTra+=1;
        }
      }
    }

ldab, 2024-02-14:
What is the value of “ Foto”?

can you elaborate what is “unredeable” and how you determined that it was transferred correctly?

ldab, 2024-02-14:
You are creating a file named with the content of “foto”… what is the char like?

biggiu, 2024-02-14:
Sorry but I don't understand what you mean.
To try to transfer the files from the SD-CARD to the FTP area, having no examples to follow, I thought of doing this:
CerFotVia:
locates the files present in the work folders and then for each file detects:
file.path() - file.name() - file.size()

Then I pass these parameters to the function:
TraFotVia .path() and .size()

I convert the string to CHAR
   w0 = FotName.substring(1);
   LW = w0.length()+1;
   w0.toCharArray(Photo, LW);

I write to FTP

ftp.WriteData( (unsigned char *)Photo,sizeof(Photo));

I don't know if this is the correct procedure because in your sketch:

ftp.WriteData( octocat_pic, sizeof(octocat_pic) );

there is no example of source file and destination file.

Maybe my procedure is wrong but I don't know how else to do this transfer.

ldab, 2024-02-14:
When you try to open the file on your graphic program, what is the file name?

biggiu, 2024-02-14:
Sorry but I'm at a client's house at the moment, out of the office. In the early afternoon I will send you the image.

biggiu, 2024-02-14:
0355-20240212171954.jpg is ok
0355-20240212171954-T.jpg 

[image]
[image]

ldab, 2024-02-14:
> 0355-20240212171954.jpg is ok 0355-20240212171954-T.jpg
> 
> ![0355-20240212171954-T](https://private-user-images.githubusercontent.com/52693747/304772677-43bd694c-671e-4d5a-bb66-98d9f36c70a4.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5NDAyNTUsIm5iZiI6MTcwNzkzOTk1NSwicGF0aCI6Ii81MjY5Mzc0Ny8zMDQ3NzI2NzctNDNiZDY5NGMtNjcxZS00ZDVhLWJiNjYtOThkOWYzNmM3MGE0LkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE0VDE5NDU1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3YTA5NWU2NGY1MGJhNGEyN2E4ODQ5NWU3ZWNiMmI5ZjQyZDI5MGQ1MmE1ODNlY2YyOTZjMzZhNmVmMGRjNjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.NsP-5UMyX26W0vbTEFn-rbFTSCrRI85o-4P_Mi790hk) ![0355-20240212171954](https://private-user-images.githubusercontent.com/52693747/304772672-a45adcc7-cce8-4c0b-80d2-3ca7a9822dd1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5NDAyNTUsIm5iZiI6MTcwNzkzOTk1NSwicGF0aCI6Ii81MjY5Mzc0Ny8zMDQ3NzI2NzItYTQ1YWRjYzctY2NlOC00YzBiLTgwZDItM2NhN2E5ODIyZGQxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE0VDE5NDU1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA4NzU2Y2ZlY2QxYjRmN2U2NWY2N2YzNTNlYmE4M2U2YTRiOTU3MmUzZjc1MzNlOWY0MmFjY2I1YmI4YTJlMDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.rzQkBjOO46_qm6lLeVYYi4hP4p0A2JnfLVSSurWzf58)

Try to open the file you believe you uploaded correctly with a text editor, you will find the problem there, ex.

```
00355/0355-20240212171954.jpg û?# # 150 Accepted data connection
 205 40 38 195 160)
 em - No anonymous login
220 You will be disconnected after 15 minutes of in´û?ø
@?    è      ý?`½ý?€­ý?¨­ý?  ˆ      ø
@?    è      ”®ý?ªý?°ªý?Øªý?   ˆ         p_ü?_ü?Ô_ü? 00355/0355-20240212171954.jpg û?# # 150 Accepted data connection
 205 40 38 195 160)
 em - No anonymous login
```

closed 2024-02-14 by ldab