ldab/ESP32_FTPClient #47: Upload image from SD-CARD

biggiu, 2024-02-15:
"Try to open the file you believe you uploaded correctly with a text editor, you will find the problem there, ex."

Sorry but I don't understand what you mean.
Where am I wrong in transferring the image?
What is the correct way to transfer images from SD-CARD to FTP?
When you developed the very functional library, you will have foreseen this option. I would therefore ask you to explain it to me too so that I can use it. If the option is not provided, then just know and I will adapt to find a different solution.

Thank you for the time you dedicated to the problem, I think it will help everyone to grow in their knowledge of the potential of the library.

And in any case, it would be kind of you to avoid closing the conversation before having new feedback.

ldab, 2024-02-15:
> "Try to open the file you believe you uploaded correctly with a text editor, you will find the problem there, ex."
> 
> Sorry but I don't understand what you mean.

If you open the file with the text editor you will find string on the file, the FTP responses, meaning that you are saving incorrect content to the image file. Probably the way you parse and save the file content is wrong.

> Where am I wrong in transferring the image? What is the correct way to transfer images from SD-CARD to FTP? 
There are many ways, I will not look into your code, so can't help you with that.

biggiu, 2024-02-15:
Thanks again for your kind cooperation.
I'm not asking you to look at my code, I'm asking you to give me some links that will help me understand how to do it.

ldab, 2024-02-15:
> I'm asking you to give me some links that will help me understand how to do it.

Sorry, I will not be able to help you with that. If you believe there is an issue with the library feel free to open another issue.

closed 2024-02-15 by ldab